32blit/32blit-sdk #1: Investigate SD-card IO speed

Gadgetoid, 2019-11-27:


Gadgetoid, 2020-01-22:
I think https://github.com/pimoroni/32blit-beta/commit/8c1c1bb63184bc30d229c6428c2f5ca96e8ab0f7 finally brought SD card performance up to reasonable levels.

However I have personally seen issues running the SPI interface with `SPI_BAUDRATEPRESCALER_4` on some SD cards versus `SPI_BAUDRATEPRESCALER_8` which brings the speed, in theory, back down to rates prior to this fix.

Some method of detecting the SD card, trying `SPI_BAUDRATEPRESCALER_4` and failing back to `SPI_BAUDRATEPRESCALER_8` would be great for best performance, but for now we might need to drop to `SPI_BAUDRATEPRESCALER_4` in order to maximise compatibility and minimise "why isn't my SD card working?" questions.

Daft-Freak, 2020-02-03:
I don't know what the clock with _4 is, but if it's > 25Mhz, you need to switch the card into "high speed" mode first. Also, you're meant to init with a clock of 400kHz.

(I dug up my notes from when I was attempting to get SD cards working.)

PR #323 (2020-08-14): Make SD cards fast
Replaces the SPI functions for accessing the SD card with simpler more direct versions. Improves read speed by up to 7x on my new card and ~2x on my old one.

Possibly needs some testing on more cards, but doesn't cause the cards I tested with to stop working. I've also tested that mass storage still works, since it was previously broken by changes here. (It isn't any faster though...)

[test bins](https://github.com/pimoroni/32blit-beta/files/5074139/sd-test.zip)

Before/After:
[image]

Fixes #1?
closed 2020-08-14 by Gadgetoid