32blit/32blit-sdk #6: Some of the example can't be build

Godzil, 2019-11-27:
It's probably to be expected considering it's an early release, but some of the example can't be build as is:

- [ ]  fire
 Complain about missing `engine` and `graphics` namespace (they don't exist in the 32blit source tree)
- [ ] flight
 For similar reasons (there are more errors, but could be due to the namespace issue)
- [ ] matrix-test
 Similar reasons again
- [ ] platformer
 Similar reasons, also look for an `input` namespace
- [ ] sprite-test
 Similar reasons
 - [ ] logo
 Again same reasons
- [ ] particles
 Same
- [ ] tilemap_test
 Same
- [ ] racer
 This one is an oddball. It's at linking that error come, it miss the functions `update(unsigned int)`, `render(unsigned int)` and `init()`

All the non mentioned one seems to compile.

Gadgetoid, 2019-11-27:
Ooof, good catch. I'd only ported a couple of examples to the "new" API since it was still in flux and I didn't want to waste effort.

I've now ported everything to the "new" API, albeit `racer` is an empty file- I should probably put some sort of TODO in there- and `tilemap_test` is currently segfaulting for some reason.

ali1234, 2019-11-28:
`flight` segfaults because spritesheet::generate_mimaps() is causing random memory corruption.

Gadgetoid, 2020-01-08:
I just built `rotozoom` and it crashes on startup in Windows. It's possibly doing something a little untoward.

closed 2020-03-02 by Gadgetoid