32blit/32blit-sdk #12: Joystick X axis

ali1234, 2019-12-01:
Based on the `geometry` example, it looks like joystick X value goes positive when you push left. This is the opposite of screen coordinates. This assumes the example is not buggy, which I currently have no way of verifying.

If this is the case, I strongly recommend flipping the joystick X axis in the STM HAL to make it easier to understand.

Joystick Y axis seems to go positive when you push down, which matches screen coordinates.

Note: In SDL, right and down are positive, so one of the HAL will need to handle this no matter what.

Gadgetoid, 2019-12-02:
In the STM32 firmware right is currently positive (counter to the SDL) and down is negative.

I'd tend to agree- in both cases the joystick direction should match screen coordinates and, indeed, SDL's take on things.

Gadgetoid, 2019-12-02:
Testing in the SDL environment and the joystick matches expectations (using an Xbox One controller). I believe the STM32 HAL should match this.

Since joystick input is currently a vec2, it might make sense to scale it to -+1.0 in both cases since that would make using the joystick input to scale movement or rotation values very easy.

Gadgetoid, 2019-12-03:
Added joystick support to the raycaster demo with a smidge of deadzoning, works really well with my Xbox One controller: https://github.com/pimoroni/32blit-beta/commit/8912540fb35dae6e0d3a4cf843677622dada70ba

Also fixed support in geometry, but there's still work to do there: https://github.com/pimoroni/32blit-beta/commit/5d096474d07ca4e04914330a4c56430538792cbd

Gadgetoid, 2019-12-06:
I've fixed the STM32 HAL to match SDL, so joystick input will be consistent across platforms: 972b02084094c2b66ac15b4ab970b01d18779e1c

Gadgetoid, 2019-12-10:
I'd somehow broken the Y-axis in an earlier change and fixed again in https://github.com/pimoroni/32blit-beta/commit/c4a56e868acb43fd9783bfa769a8b069e2f6cfb7#diff-690b2c1f2e6fcd12a29bc1865bc4af1fR307

closed 2020-03-02 by Gadgetoid