32blit/32blit-sdk #13: Asset-Builder does not accept rgba png files

nerathul, 2019-12-02:
tried to get a rgba png file formated from aseprite to use in code, get error from line 207 -> too many arguments, expected 3.
[image]
[image]

Gadgetoid, 2019-12-02:
I've updated `asset-builder` to attempt to paletize RGB and RGBA images. Since RGB and RGBA output formats are currently not supported, this means that any RGB or RGBA image that uses 255 or fewer colours will be crunched to an indexed palette format which is currently the one we use in most (all?) examples. This represents a significant size decrease, since RGBA and RGBA are 32bpp and 24bpp respectively whereas paletted is at most 8bpp.

Note: the `--format` parameter currently does nothing, and was indended to control the output format between unpacked RGB/RGBA and paletted.

LordEidi, 2019-12-20:
The problem I noticed using the asset-builder is that the PNG (with alpha) will get a format of RGB565 (2) in the generated C file. If you try to load that, the engine will crash with a SIGSEGV in surface.cpp on line 32. I noticed that the bounds the line before are wrong (0,0).

When I change the format to 3 (P) by hand and then run the app, everything runs fine (although the coordinates are strange, but that is a different problem).

Gadgetoid, 2020-03-02:
Closing, since `asset-builder` and `sprite-builder` are deprecated in favour of https://github.com/pimoroni/32blit-tools

closed 2020-03-02 by Gadgetoid