32blit/32blit-sdk #18: Namespace pollution

ali1234, 2019-12-08:
The following engine headers include `using namespace blit;`.

- [ ] types/map.hpp
- [ ] graphics/surface.hpp
- [ ] graphics/tilemap.hpp

This causes every file that includes them to also use the namespace, leading to random name collisions in user code. This filters through to any file that includes `32blit.hpp`.

Note that fixing this will probably break all the examples that don't already contain `using namespace blit;`.

Gadgetoid, 2019-12-09:
Let it be said that I broadly hate the use of `using namespace blit;` and would probably prefer everything to use the longform, namespace-qualified names. It leaves much less opportunity for name conflicts, or shadowing? Is that even a thing in C++. Although I can't say something if like `colour = blit::colour(0,0,0);` is more or less desirable.

I guess I can't shake that it reminds me of one of my most-hated Python code-smells: `import numpy as np`.

ali1234, 2019-12-09:
The issue here is using it inside headers - it's fine if you do it inside the .cpp because it will never affect user code.

And yes name shadowing most certainly is a thing. I have a variable called buttons and the IDE complains that it shadows blit::buttons everywhere.

commit f539feed (2020-01-16): Remove remaining using namespace blit in headers

Fixes #18
closed 2020-01-16 by Gadgetoid