32blit/32blit-sdk #19: Defining button bit masks as enum is a bad idea

ali1234, 2019-12-08:
`engine/input.hpp` defines the bit masks for each button in an enum:

```
  enum button {
    DPAD_LEFT = 1,
    DPAD_RIGHT = 2,
    DPAD_UP = 4,
    DPAD_DOWN = 8,
    A = 16,
    B = 32,
    X = 64,
    Y = 128,
    MENU = 256,
    HOME = 512,
    JOYSTICK = 1024
  };

  extern uint32_t buttons;
```

The type of an enum is undefined - it may be signed or unsigned. Usually is it signed. If you try to take the bitwise or of buttons and a button mask defined in the enum, this gives a compiler warning.

I think this should be exactly equivalent (in particular, it should use no memory):

```
  namespace button {
    static const uint32_t DPAD_LEFT = (1<<0);
    static const uint32_t DPAD_RIGHT = (1<<1);
    static const uint32_t DPAD_UP = (1<<2);
    static const uint32_t DPAD_DOWN = (1<<3);
    static const uint32_t A = (1<<4);
    static const uint32_t B = (1<<5);
    static const uint32_t X = (1<<6);
    static const uint32_t Y = (1<<7);
    static const uint32_t MENU = (1<<8);
    static const uint32_t HOME = (1<<9);
    static const uint32_t JOYSTICK = (1<<10);
  }

  extern uint32_t buttons;
```

tinwhisker, 2019-12-09:
Would `enum button : unsigned int { ... };` not read better, or not valid here?

ali1234, 2019-12-09:
I believe typed enums are a C++11 feature, so that should be okay.

ali1234, 2019-12-09:
Note there is a difference between doing it with `enum` vs `namespace`: The enum also puts the names into the parent scope. So with enum you can do both `blit::DPAD_LEFT` and `blit::buttons::DPAD_LEFT` but with namespace you can only do `blit::buttons::DPAD_LEFT`.

Gadgetoid, 2019-12-10:
A quick test of applying an `unsigned int` type to the enum suggests that it still compiles and functions as expected. I've pushed it for now, pending further discussion.

closed 2020-01-08 by Gadgetoid