32blit/32blit-sdk #21: 32blit - Improve and relocate text drawing

Gadgetoid, 2019-12-11:
Presently `surface::text` lives in `surface.cpp` and defines a very crude text drawing function that's in need of some TLC: https://github.com/pimoroni/32blit-beta/blob/666c5d2ef0315981b02c5eaf448ada026c09fa43/32blit/graphics/surface.cpp#L446-L516

1. Needs to support alignment within its defined `rect()`
2. Needs to support clipping to screen bounds, and potentially also wrapping/clipping to `rect()` bounds
3. Needs relocating into `text.cpp` for better separation of concerns

Right now the use of syntax like `&minimal_font[0][0]` to address the first byte of a specific font is ungainly. Font should possibly be a struct with character size and data address or something similar.

One would expect something like the following:

```
fb.text("Hello World", blit::font::minimal, rect(0, 0, SCREEN_WIDTH, SCREEN_HEIGHT), blit::text_align::center_center);
```

To place the text "Hello World" in the center of the display.

And

```
fb.text("Hello World", blit::font::minimal, point(100, 100), blit::text_align::center_center);
```

To place the text centered vertically and horizontally around the given screen coordinate.

Gadgetoid, 2019-12-11:
A python example of a text-in-rect reflow function:

https://github.com/pimoroni/pidi-plugins/blob/4ec06516c5e56433eff384a491b5a7c5252d3774/pidi-display-pil/pidi_display_pil/__init__.py#L15-L53

`line_spacing` is an interesting point here- since we'd expect text in a `rect()` to wrap automagically and may want to control this.

ali1234, 2019-12-11:
"wrap" should probably be an option (possibly one that gets implemented much later).

One of the key things needed to make all this work is a function that takes a string and a font and returns the pixel size it would be rendered at. This is easy for fixed width fonts, it is just `strlen * character width`. But for variable width you have to go through the string and calculate it. Note that currently even if you ask for fixed width text, spaces are a different width to the rest of the characters.

Daft-Freak, 2020-01-10:
I made an attempt at improving this (I had some similar code in another project): https://github.com/Daft-Freak/32blit-beta/commits/text (Likely needs a little cleanup).

Gadgetoid, 2020-01-21:
@Daft-Freak would love that in a pull request! Looks extremely comprehensive and certainly considerably better than the current implementation.

closed 2020-03-02 by Gadgetoid