32blit/32blit-sdk #22: timers

nerathul, 2019-12-12:
timers do not start counting after setting duration, start and loops and do even not start if you update the timers manually

Gadgetoid, 2019-12-13:
Timers should now be in a state where they at least somewhat work, as of: https://github.com/pimoroni/32blit-beta/commit/6128e2904e96e778b31ec45964bfbb4f384f9dec

You still need to manually call `update_timers` but I'll get that fixed and an example put together ASAP.

A quick usage example:

```c++
uint32_t count;
void inc_count(blit::timer &t){
  count++;
}
blit::timer counter;
void init() {
  counter.init(inc_count, 1000, -1);
  counter.start();
}
void update(uint32_t time_ms) {
  blit::update_timers(time_ms);
}

```

Gadgetoid, 2019-12-14:
Actually you don't need to call `blit::update_timers()`- it's called by the engine `tick` function automatically for you.

closed 2020-03-02 by Gadgetoid