32blit/32blit-sdk #26: 32blit-sdl - building for win32 failing with "undefined reference to `SDL_main'"

Gadgetoid, 2019-12-23:
Using a cross-compiled SDL2 library and attempting to build a project for Win32 results in "undefined reference to `SDL_main'"

Steps to reproduce:

1. Enter any project directory
2. Make a build directory
3. Set up with `cmake .. -DCMAKE_TOOLCHAIN_FILE=../../../mingw.toolchain -DSDL2_DIR=/usr/local/cross-tools/i686-w64-mingw32/lib/cmake/SDL2 -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON`
4. Run "make"

Example verbose output:

```
[100%] Linking CXX executable platformer.exe
/usr/bin/cmake -E cmake_link_script CMakeFiles/platformer.dir/link.txt --verbose=1
/usr/bin/cmake -E remove -f CMakeFiles/platformer.dir/objects.a
/usr/bin/i686-w64-mingw32-ar cr CMakeFiles/platformer.dir/objects.a @CMakeFiles/platformer.dir/objects1.rsp
/usr/bin/i686-w64-mingw32-g++    -Wl,--whole-archive CMakeFiles/platformer.dir/objects.a -Wl,--no-whole-archive  -o platformer.exe -Wl,--out-implib,libplatformer.dll.a -Wl,--major-image-version,0,--minor-image-version,0 @CMakeFiles/platformer.dir/linklibs.rsp
/usr/local/cross-tools/i686-w64-mingw32/lib/libSDL2main.a(SDL_windows_main.o): In function `main_getcmdline':
/mnt/c/Users/32blit/Documents/Development/Pimoroni/SDL2-2.0.10/build.mingwx86/../src/main/windows/SDL_windows_main.c:175: undefined reference to `SDL_main'
collect2: error: ld returned 1 exit status
CMakeFiles/platformer.dir/build.make:102: recipe for target 'platformer.exe' failed
make[2]: *** [platformer.exe] Error 1
make[2]: Leaving directory '/mnt/c/Users/32blit/Documents/Development/Pimoroni/32blit-beta/examples/platformer/build.mingw'
CMakeFiles/Makefile2:71: recipe for target 'CMakeFiles/platformer.dir/all' failed
make[1]: *** [CMakeFiles/platformer.dir/all] Error 2
make[1]: Leaving directory '/mnt/c/Users/32blit/Documents/Development/Pimoroni/32blit-beta/examples/platformer/build.mingw'
Makefile:86: recipe for target 'all' failed
make: *** [all] Error 2
```

Main exists in `Main.cpp` of the SDL HAL, with the correct signature.

ali1234, 2019-12-23:
This is another problem of gcc requiring libraries in the right order, otherwise it removes symbols. In this case it is removing SDL_main from BlitHalSDL. We can see why by checking CMakeFiles/platformer.dir/linklibs.rsp:

   ../../32blit-sdl/libBlitHalSDL.a ../../32blit/libBlitEngine.a -L/opt/cross/i686-w64-mingw32/lib  -lmingw32 -lSDL2main -lSDL2 -mwindows -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 

CMake knows that BlitHalSDL depends on SDL2, so it puts SDL2 later in the list. However what it does not know is that BlitHalSDL contains `SDL_main` (which is really our `main`, renamed with a #define). So that symbol gets dropped because nothing needs it. This is a circular dependency - it will not work regardless of which library we put first. It is somewhat unusual that main is inside a static library, but we have little choice here. Additionally we can't use the object library trick here, because object libraries can't have dependencies.

ali1234, 2019-12-23:
We could also fix this by a few other methods, eg `-Wl,--start-group` or cmake's LINK_INTERFACE_MULTIPLICITY, but I don't think it is worth it since we always need the entire HAL library anyway.

Gadgetoid, 2019-12-23:
Tested and confirmed it produces a binary that works on Windows 10.

Looks like the resulting binaries require - from the mingw toolchain on Linux:

* `/usr/lib/gcc/i686-w64-mingw32/7.3-win32/libgcc_s_sjlj-1.dll`
* `/usr/lib/gcc/i686-w64-mingw32/7.3-win32/libstdc++-6.dll`

And, of course, from the cross-compiled SDL2 library:

* `/usr/local/cross-tools/i686-w64-mingw32/bin/SDL2.dll`

Location may vary.

I don't know what the etiquette is with cross compiling, but I've peeked into dozens of game installation directories and shipping accompanying DLLs is entirely usual.

PR #27 (2019-12-23): Sdl main fix
libBlitHalSDL.a contains SDL_main, which is required by libSDL2.a.
But libBlitHalSDL.a also depends on libSDL2.a, so we have a circular
dependency. Which ever one is listed first will have some required
symbols stripped by the linker. In order to avoid this, just use
--whole-archive when linking libBlitHalSDL.a, because we almost
certainly need all of it anyway.

This whole situation happens because it is rather unusual to have
main() inside a static library, rather than the source of the final
executable. But due to what we're doing, we doing have much choice.

Fixes #26
commit 1c6a06e3 (2019-12-23): Include the whole SDL HAL library

libBlitHalSDL.a contains SDL_main, which is required by libSDL2.a.
But libBlitHalSDL.a also depends on libSDL2.a, so we have a circular
dependency. Which ever one is listed first will have some required
symbols stripped by the linker. In order to avoid this, just use
--whole-archive when linking libBlitHalSDL.a, because we almost
certainly need all of it anyway.

This whole situation happens because it is rather unusual to have
main() inside a static library, rather than the source of the final
executable. But due to what we're doing, we don't have much choic [... cut]
closed 2019-12-23 by Gadgetoid