32blit/32blit-sdk #32: Missing graphics primitives

ali1234, 2020-01-02:
The following graphics primitives aren't implemented in the blit API. They probably should be for completeness:

 - [ ] Unfilled rectangle.
 - [ ] Unfilled circle.
 - [ ] Filled ellipse.
 - [ ] Unfilled ellipse.
 - [ ] Filled polygon.

An unfilled ellipse algorithm is described here: https://dai.fmph.uniba.sk/upload/0/01/Ellipse.pdf

Once ellipses are implemented, the circle implementations can be a wrapper around them to save space.

Filled polygons can be implemented by drawing filled triangles:

```
    for(n=1; n<(points-1); n++)
        triangle(p[0], p[n], p[n+1]);
```

ahnlak, 2020-11-26:
Having just gone to look up the unfilled rectangle API function... is this still on the 'wanted' list? :-)

AnthonyDiGirolamo, 2021-11-29:
+1 I'd like this

mikerr, 2023-02-17:
Actually  looking at this, **filled polygon** is already in the API:

`void Surface::polygon(std::vector<Point> points)`

and **outline_circle** is in [primtive.cpp](https://github.com/32blit/32blit-sdk/blob/master/32blit/graphics/primitive.cpp)  but commented out.

Gadgetoid, 2023-02-18:
I wonder if it's worth porting https://github.com/lowfatcode/pretty-poly and https://github.com/lowfatcode/alright-fonts or a little ambitious.

Daft-Freak, 2023-02-18:
:thinking: None of the examples use `polygon`, maybe we need a "primitives" example...

Got enough changes waiting for a release already though :laughing: