32blit/32blit-sdk #33: blit::rand() should not use std::rand()

ali1234, 2020-01-02:
The reason is actually explained in the commit:

> 32blit's RNG is a non-deterministic random number generator or true-RNG. Its output cannot be predicted and used to generate deterministic levels from seeds. std::rand() or perhaps your own LFSR or deterministic number sequence generator should still be used for this.

The SDL HAL uses `std::rand()` to implement `blit::rand()`. This means that if you use both of them in your game, then `std::rand()` will produce different values on hardware vs SDL.

The SDL implementation of `blit::rand()` should instead use some other implementation which does not affect `std::rand()`.

Gadgetoid, 2020-01-02:
Ooof- really good point. Thank you.

Do you have any opinions about what to use in lieu of `std::rand()`? 

Perhaps the C++11 random library which has `std::random_device`

ali1234, 2020-01-02:
Yes, C++ <random> has RNG objects which in theory don't affect each other. No idea how to use it, but it seems like the best way to go.

Gadgetoid, 2020-01-02:
Quick and dirty POC:

```
#include <random>
#include <iostream>

int main(int argc, char **argv) {
        std::random_device random_device;
        std::mt19937 random_generator(random_device());
        std::uniform_int_distribution<uint32_t> random_distribution(0, 0xffffffff);

        for (int n=0; n<10; ++n) {
                std::cout << random_distribution(random_generator) << ' ' << std::endl;
        }

        return 0;
}
```

Seeded with `random_device` to mimic the RNG, but uses a 32-bit mersenne twister PRNG for speed rather than continually relying on the system as a source of randomness.

closed 2020-01-08 by Gadgetoid

peteroupc, 2020-01-20:
The `random_generator` here is improperly seeded; see "[C++ Seeding Surprises](http://www.pcg-random.org/posts/cpp-seeding-surprises.html)" by @imneme ; see also "[Developing a seed_seq Alternative](http://www.pcg-random.org/posts/developing-a-seed_seq-alternative.html)".