32blit/32blit-sdk #47: Proposal: Split README.md into OS-specific and general setup

Gadgetoid, 2020-01-08:
Right now our README.md is getting bigger by the minute with all the necessary steps to get working.

While the Wiki is great for FAQs and more esoteric issues I think it might be worth creating a `documentation` folder and having a `WINDOWS.md`, `LINUX.md` and `OSX.md` for noise-free instructions.

Thoughts?

Gadgetoid, 2020-01-08:
I see @ntwyman @lenardg @LordEidi @johnmccabe and @ali1234 have contributed to the readme so far (which is awesome in and of itself) so your input on this would be the most valuable. Thank you!

lenardg, 2020-01-08:
This same thought crossed my mind while writing the WSL content. So I am in favor of this!

But I would call the folder just simply `docs` :)

Gadgetoid, 2020-01-08:
As of https://github.com/pimoroni/32blit-beta/commit/691a68b6f3aaa94f9430a224a5f5c94ce4df4766 I've also poured Emscripten onto this doc-fire, so it's getting out of hand!

LordEidi, 2020-01-09:
I'd prefer a /doc folder with multiple .md files in it over the wiki for a simple reason. You can download (well git clone) the whole /doc folder and have things locally. Doing so with the wiki is not as easy. And yes I'd go with /docs or /doc vs /documents.

Which also shows my preference regarding a single vs multiple files. Anything we could help there with?

Gadgetoid, 2020-01-10:
Any suggestions for making the documentation clearer and more complete are always welcome. I think each new person who sets up and runs into issues helps iron on the problems.

Should probably - and I'm leaving this note to myself more than anyone - get some details of the Windows 10 DFU Mode driver issue in there too- whereupon 32blit seems to appear as "STM32 In FS Mode" rather than "STM32 In DFU Mode" and doesn't show up in DfuSe.

closed 2020-12-30 by Gadgetoid