32blit/32blit-sdk #56: Proposal: replace all linear algebra code with linalg.h

ali1234, 2020-01-12:
I recommend replacing all linear algebra code with this header:

https://github.com/sgorsten/linalg

It is public domain and very complete. It is all implemented using templates so it supports any type.

For example using linalg.h, vec2.hpp can be completely replaced by the following:

```
#include <linalg.h>
typedef linalg::vec<float, 2> vec2;
```

and perhaps a few syntactic changes where vec2 is used.

Gadgetoid, 2020-01-12:
We’ve a few reservations about replacing the existing code with template-driven functions, since:

* they make it harder to predict what impact code might have on the binary size
* templates make for some of the most opaque and confusing C++ errors for beginners
* it’s too easy to do something unintentional like accidentally leak doubles all over the codebase - the template will handle this gracefully and slow down the code
* relying on a third party library may close the door to future blit-specific use of SIMD or FPU instructions for optimisation

Writing our own functions may seem bonkers, but it makes the API simpler and more suited to Blit’s original target- a limited embedded system.

We’ll definitely look over this to find gaps in our API- and encourage anyone advanced enough to use this library in lieu of ours.

There are good cases for the use of this library to be sure- but I don’t think it aligns with our intentions for the Blit API at the moment.

ali1234, 2020-01-12:
I'm not really convinced by any of these reasons.

- Code size will be unaffected because everything is inline in both cases, but use of `constexpr` will probably end up reducing code size since more can be calculated at compile time.

- Templates have the same promotion rules as regular code, but use of `explicit` makes it more strict than the current implementation. Either way, there is no way to accidentally promote a vector type and the templated functions won't let you mix them. This won't compile:

```
linalg::vec<float, 3> v;
linalg::mul(v, 3.0);
```
But this will:
```
linalg::vec<float, 3> v;
linalg::mul(v, 3.0f);
```

- FPU will be used automatically by the standard library when compiling with `-ffast-math -fno-math-errno` and Cortex M7 has no float SIMD.

- Filling in the gaps of the current implementation will be quite a task because there is so much missing from it. You'll either end up with the same thing, or 96 overloaded versions of the same thing if you try to do it without templates.

That just leaves the error message problem, but that is just the nature of C++ and I don't think there is anything you can do about that.

closed 2020-03-02 by Gadgetoid