32blit/32blit-sdk #59: fb.clear() should reset the pen to black

lenardg, 2020-01-13:
Currently fb.clear() does not reset the pen to black. It will clear the screen with whatever pen was last selected in the render method. 

So this render method:

``` C++
    fb.clear();
    fb.pen(blit::rgba(255, 255, 255));
    fb.rectangle(rect(5, 5, 5, 5));
```

Will actually paint the entire screen white from the second time rendering is called. The tiny rectangle will not be visible at all, because everything is white.

I recommend `clear()` should reset the pen to black as well. 

Maybe even add a new `clear_color` rgba variable that would be black by default, and that can be changed to whatever you want your clear color to be.

ahnlak, 2020-01-13:
It clears the screen *to the current pen colour*, in keeping with all the other graphics primitives.

I don't think it should be prescribing what colour is 'clear'.

mikerr, 2020-01-13:
Normally a clear command would clear in the current _background_ colour,
but since 32blit API currently only has foreground (pen) colour defined, it's good to continue using that.

lenardg, 2020-01-13:
@mikerr good point there, maybe I would then like to recommend we also have some sort of brackground colour as well, and that it would be black by default. 

I would rephrase my original problem like so: I find it counterintuitive that my background is cleared with the same colour that I am using to draw onto the screen.

trollied, 2020-01-13:
I think a `    fb.background(rgba(255, 255, 255));` makes most sense.

ahnlak, 2020-01-13:
My only thought on that is what else would the background be used for? All the other primitives effectively draw with a 'transparent' background, and I'm not sure it would make much sense to change that.

Perhaps the problem is calling it `clear()` rather than something more 'painty' (perhaps `fb.fill()`?)

descartes, 2020-01-26:
Perhaps have `clear()` take an optional colour rgba triplet rather than have a `fb.background` for that one situation - effectively combining the two in to one.

Gadgetoid, 2020-02-13:
We've refactored `screen.pen(color)` into `screen.pen = color` so the actual run-time overhead of just setting the pen colour before calling `clear()` is minimal.

This is something we'll cover in the documentation/tutorials and I think users will see enough of:

```
screen.pen = Pen(0, 0, 0)
screen.clear()
```

To understand the relationship.

`clear()` is really just a lazy alias for `rectangle(screen.bounds)` and since all of our API is stateful, rather than accepting arguments it doesn't make sense to make it `screen.clear(color)`.

Right now- for better or worse- we have no concept of foreground/background/stroke/fill colours and there's no sense changing that for a single case.

closed 2020-03-02 by Gadgetoid