32blit/32blit-sdk #63: Improve button handling in the API

lenardg, 2020-01-13:
Currently the API provides us with button statuses only (is it pressed or is it released). The demo game scrolly-tile does some bitfield magic to keep track of what is actually done with the buttons - and not just if it is held or not. 

Relevant code:
``` C++
	uint16_t changed = blit::buttons ^ last_buttons;
	uint16_t pressed = changed & blit::buttons;
	uint16_t released = changed & ~blit::buttons;
```

It would be great if this same functionality would be part of the engine. 

Instead of just `blit::buttons` it could be a struct of some kind, where you would have three members:

* `status` - to check the status of a single button, like what blit::buttons is now.
* `pressed` - that would tell which button was pressed since last time
* `released` - that would tell which button was released since last time

ali1234, 2020-01-15:
You can avoid the complement operation:

    uint16_t released = changed & last_buttons;

Daft-Freak, 2020-08-28:
I think I did (quite close to) this in #310?

closed 2020-10-01 by Gadgetoid