32blit/32blit-sdk #70: Tools readme file containing new Sprite Builder description

LordEidi, 2020-01-17:
I messed up my branches and can't seem to do a pull request without handing over my first born.

Please copy the content from my fork here (https://github.com/LordEidi/32blit-beta/blob/master/docs/Tools.md) and add it to the docs folder. You might want to remove the Readme.md in the tools folder for it still points to the old script.

Gadgetoid, 2020-01-17:
Messing up branches- story of my life :D Have created #71 to action this- let me know if it looks okay and I'll merge!

LordEidi, 2020-01-17:
Oh well, this happens if you usually use the Git integration in your IDE and then mix that with some CLI magic... Again, thanks for merging in an unconventional way.

closed 2020-01-17 by LordEidi