32blit/32blit-sdk #74: Flight example fails to build for device

Daft-Freak, 2020-01-21:
```
[build] [1/1 100% :: 0.066] Linking CXX executable examples/flight/flight.elf
[build] FAILED: examples/flight/flight.elf 
[build] : && /usr/bin/arm-none-eabi-g++  -mcpu=cortex-m7 -mthumb -mfloat-abi=hard  -mfpu=fpv5-d16 -Os -Wall -fdata-sections -ffunction-sections -Wattributes -Wdouble-promotion -Wno-unused-variable -Wno-write-strings -fpermissive -Os -DNDEBUG  -specs=nosys.specs -Wl,--gc-sections -specs=nano.specs -T /home/daftfreak/programs/32blit-beta/32blit-stm32/STM32H750VBTx_FLASH.ld    -Wl,-Map=flight.map,--cref 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/main.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/stm32h7xx_it.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/stm32h7xx_hal_msp.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_adc.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_adc_ex.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_cortex.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_dac.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_dac_ex.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_hrtim.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_smbus.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_ltdc.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_ltdc_ex.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_dsi.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_qspi.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi_ex.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_tim.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_tim_ex.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_rcc.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_rcc_ex.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_flash.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_flash_ex.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_gpio.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_hsem.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_dma.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_dma_ex.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_mdma.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_pwr.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_pwr_ex.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2c.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2c_ex.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_exti.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_rng.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_pcd.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_pcd_ex.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Middlewares/Third_Party/FatFs/src/diskio.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Middlewares/Third_Party/FatFs/src/ff.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Middlewares/Third_Party/FatFs/src/ff_gen_drv.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Middlewares/Third_Party/FatFs/src/option/syscall.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_core.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ioreq.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/system_stm32h7xx.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/user_diskio.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/fatfs.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/fatfs_sd.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/usb_otg.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/usb_device.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/usbd_conf.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/usbd_desc.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/usbd_cdc_if.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/gpio.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/adc.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/dac.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/debug.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/dma.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/hrtim.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/i2c.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/i2c-msa301.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/i2c-bq24295.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/ltdc.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/quadspi.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/spi.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/spi-st7272a.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/sys.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/tim.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/rng.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/32blit.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/CDCLogging.c.obj 32blit-stm32/CMakeFiles/BlitHalSTM32.dir/startup_stm32h750xx.s.obj examples/flight/CMakeFiles/flight.elf.dir/flight.cpp.obj  -o examples/flight/flight.elf  32blit/libBlitEngine.a && cd /home/daftfreak/programs/32blit-beta/build/examples/flight && /usr/bin/arm-none-eabi-objcopy -O ihex /home/daftfreak/programs/32blit-beta/build/examples/flight/flight.elf flight.hex && /usr/bin/arm-none-eabi-objcopy -O binary -S /home/daftfreak/programs/32blit-beta/build/examples/flight/flight.elf flight.bin && /home/daftfreak/programs/32blit-beta/tools/dfu build --force --out flight.dfu flight.bin && arm-none-eabi-size /home/daftfreak/programs/32blit-beta/build/examples/flight/flight.elf
[build] /usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: examples/flight/flight.elf section `.rodata' will not fit in region `FLASH'
[build] /usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 5528 bytes
```

Looks like #38 increased code size enough to make this example too big.

ali1234, 2020-01-21:
The FAT filesystem code adds more to the binary than the CDC stuff. You need to disable them both to get binaries back down to the previous size. The FAT code now always gets linked because the audio test code looks for a file on the SD card before running the user code.

I had to disable both of them to fit my code: https://github.com/pimoroni/32blit-beta/commit/f8993016e8515f2239b7a7c08df5d3db1e841228#diff-255e6dbf7317553d0408b7fce0abf029

Daft-Freak, 2020-01-21:
Hmm, yeah FAT code looks quite big. I mentioned #38 since I bisected the build failure back to around there (there's a lot of commits that don't build).

Daft-Freak, 2020-02-02:
The good news is that flight now builds since the external flash is used now. The bad news is that it doesn't run.

mylogon341, 2020-02-04:
@Daft-Freak When you say 'doesn't run', how do you mean? I made quite a few changes to this example project and have it running fine in the emulator then was quite gutted when it didn't run on device - what I experienced was a checkered screen. I couldn't get the menu open so had to force the device into dfu mode to load something else on to it.
Is the same thing you have experienced?

Daft-Freak, 2020-02-04:
Device hangs with mostly black screen and some corruption(?) in the top left. Haven't tried to debug (mostly because I can't debug on device).

Daft-Freak, 2020-02-04:
Significant digging later...

Since 604d8f05394856d60fd123ee2d6f91ab97bba7eb, the mode7 code attempts to blend an RGBA sprite to an RGBA surface, and ends up calling a null function pointer here 
https://github.com/pimoroni/32blit-beta/blob/b83983d6b476a4e8161f2d06a1ef3381eeebfa53/32blit/types/map.cpp#L83

since that entry (0) is null
https://github.com/pimoroni/32blit-beta/blob/b83983d6b476a4e8161f2d06a1ef3381eeebfa53/32blit/graphics/surface.cpp#L36-L39.

Daft-Freak, 2020-03-07:
Builds fine now, mode7 is just generally broken/disabled.

closed 2020-03-07 by Daft-Freak