32blit/32blit-sdk #89: render() & update() time

HeadBoffin, 2020-01-29:
I see a comment from @Gadgetoid in the Engine/API project cards about sending deltas rather than absolute time.

I can easily calculate the deltas but conversely I could accumulate them to get the actual time - however my brain says it would like the actual time so if I want something to happen as soon after a certain amount of time has elapsed, we can store time + x somewhere and quickly check that at the beginning of either block.

I'm aware of the timer API but I'm thinking of keeping things simple for novice programmers (me, school students, etc) and I sort of like the init/render/update setup as a mirror of the constructs they see (micro:bit on-start & forever or Arduino setup & loop)

lowfatcode, 2020-01-29:
The way that time is passed into `update` is probably not quite how you expect. Basically we guarantee you will get one `update` call for every 10ms of execution, however depending on how long your `render` and other processing takes that may mean you get a flurry of calls to catch up. 

Anything based on "absolute time" should use the `time_ms` parameter as the truth.

See the section titled "Play catch up" on this page for more information: http://gameprogrammingpatterns.com/game-loop.html#play-catch-up

The reason we chose this approach is that it is a lot easier for beginners to work with, it means every update step is exactly the same time period so you don't need to scale every action differently each time. For example if you want your character to move left 1 pixel every 10ms then you can just do something like `player.position.x--;` in your `update()` function. This has big benefits for functionality like movement, collision detection, outputting sound consistently, etc.

HeadBoffin, 2020-01-29:
> The way that time is passed into `update` is probably not quite how you expect. 

OK, calls at 10ms intervals - is there a limit to the backlog?

> Anything based on "absolute time" should use the `time_ms` parameter as the truth.

So the time passed in to update **is** actually the absolute time, I just may get one or more calls as soon as I finish.

My test code dumped the time to the debug output so I could get a feel for what happened when I did things - not the same as the actual device, but I suspect my i7 iMac is faster - and could see that the calls came thick and fast but not always on a locked down schedule.

So probably best to put in a check if we've done a chunk of work that we aren't immediately updating / re-rendering, thereby growing the two queues and eventually disappearing up our own stack. I'm thinking of procedurally generated maps on a scene change for instance.

lowfatcode, 2020-01-29:
> OK, calls at 10ms intervals - is there a limit to the backlog?

There is currently no limit to the backlog but we'll likely add a sanity check in future that shows an error if the backlog grows too far.

> So the time passed in to update is actually the absolute time, I just may get one or more calls as soon as I finish.

The time passed into `update` is an "artificial" time for that specific iteration of the update loop - it is the only time value you really should be basing any time related functionality against. :-)

closed 2020-01-30 by lowfatcode