32blit/32blit-sdk #91: STM32 build fails

perses-games, 2020-01-29:
STM32 build fails (at least on Linux). There is the following error:

```bash
rnentjes@rnentjes-MS-7B79:~/Development/32blit/32blit-beta/examples/palette-cycle/build.stm32$ make
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rnentjes/Development/32blit/32blit-beta/examples/palette-cycle/build.stm32
[ 73%] Built target BlitHalSTM32
[ 98%] Built target BlitEngine
[ 99%] Linking CXX executable palette-cycle.elf
Building palette-cycle.bin
Building palette-cycle.dfu
/usr/bin/objcopy: Unable to recognise the format of the input file `/home/rnentjes/Development/32blit/32blit-beta/examples/palette-cycle/build.stm32/palette-cycle.elf'
make[2]: *** [CMakeFiles/palette-cycle.dir/build.make:236: palette-cycle.elf] Error 1
make[2]: *** Deleting file 'palette-cycle.elf'
make[1]: *** [CMakeFiles/Makefile2:74: CMakeFiles/palette-cycle.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

There is a semicolon in an assignment in 32blit-stm32/Makefile:91 which shouldn’t be there. This causes the PREFIX to not be set and then it will pick up the command from the path.

Daft-Freak, 2020-01-29:
I think this needs CMAKE_OBJCOPY to be set in 32blit.toolchain. (It looks like you're using CMake, which means those makefiles aren't used).

perses-games, 2020-01-29:
@Daft-Freak I don't know much about make/cmake toolchains, I just looked where the objcopy command was used and this seemed to fix my problem.

Ah, it looks like it was using the files from the previous sdl build and all that was needed was a clean regeneration of the build files.

closed 2020-01-29 by perses-games