32blit/32blit-sdk #93: Unreliable SD Card Initialisation

Gadgetoid, 2020-01-30:
Right now I'm not 100% sure if this is a hardware fault due to me using my SD card slot as a ground connection for an oscilloscope, or an issue with how/when the SD card is initialised.

It seems to be the case that I can insert an SD card and have it work fine through device resets, but as soon as it is removed, or the battery is disconnected it will fail to work until I undergo an arcane ritual of slowly inserting the SD card while repeatedly resetting.

A vague guess is that something is preventing the card from entering SPI mode during the normal start-up procedure, and this can be fudged by not-quite-inserting-the-card. It looks like SPI mode requires the CS signal to be asserted (active low) during start up. Since the card will not exit SPI mode until a power cycle (which reset does not cause) it will work fine until power is otherwise lost.

Gadgetoid, 2020-02-06:
It looks like dropping the SD card SPI interface speed from a 4X prescalar to an 8X prescalar has a huge (positive) impact on compatibility. I've made this change temporarily, pending a more correct fix to the SD card init.

Right now the prescalar change will half SD card speed (more or less) and it really feels quite sluggish. Since I've experienced the faster speed by jockying my SD card as described above, I'm confident we'll find a way to get back to 4X generally.

Daft-Freak, 2020-02-06:
Not entirely correct speed change: https://github.com/Daft-Freak/32blit-beta/commit/1079793d99846c8924430ee3902778d5e3f83789

Daft-Freak, 2021-05-28:
Probably don't need to switch back to the faster prescaler now that the SPI code is not awful. Hmm, the inefficiency there may have been the only reason that worked as it's waaaaay out-of-spec (120Mhz!).

Daft-Freak, 2023-01-31:
I'm going to say that this was "fixed" by the various SD card optimisations making the faster clock unnecessary... and also that this issue was that clock being too fast, so reducing it wasn't a workaround, but correct.

closed 2023-01-31 by Daft-Freak