32blit/32blit-sdk #102: "'copysign' is not a member of 'std'"

Happy-Ferret, 2020-02-01:
When attempting to compile under WSL (using the ``32blit.toolchain`` file), the following error is returned.

```
/home/username/32blit-beta/32blit/graphics/mode7.cpp: In function 'vec2 blit::world_to_screen(vec2, float, float, vec2, float, float, blit::rect)':
/home/username/32blit-beta/32blit/graphics/mode7.cpp:49:24: error: 'copysign' is not a member of 'std'
     float theta_sign = std::copysign(1.0f, theta);
                        ^
/home/username/32blit-beta/32blit/graphics/mode7.cpp:49:24: note: suggested alternative:
In file included from /home/username/32blit-beta/32blit/graphics/mode7.cpp:6:0:
/usr/include/newlib/math.h:266:15: note:   'copysign'
 extern double copysign _PARAMS((double, double));
               ^
32blit/CMakeFiles/BlitEngine.dir/build.make:374: recipe for target '32blit/CMakeFiles/BlitEngine.dir/graphics/mode7.cpp.obj' failed
make[2]: *** [32blit/CMakeFiles/BlitEngine.dir/graphics/mode7.cpp.obj] Error 1
CMakeFiles/Makefile2:85: recipe for target '32blit/CMakeFiles/BlitEngine.dir/all' failed
make[1]: *** [32blit/CMakeFiles/BlitEngine.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

I'm using an Ubuntu 16.04 based WSL and gcc/g++-9

Daft-Freak, 2020-02-01:
What about `arm-none-eabi-gcc` version? Might be a little old. (Ubuntu 18.04 has 6.3.1, which works but with a pile of warnings.)

Happy-Ferret, 2020-02-01:
> What about `arm-none-eabi-gcc` version? Might be a little old. (Ubuntu 18.04 has 6.3.1, which works but with a pile of warnings.)

You're right. That one is stuck on 4.9.3

Any idea how to upgrade it under Ubuntu? Any alternative package repository?

Happy-Ferret, 2020-02-01:
I installed a new toolchain manually.

Now I'm facing this, however:

```
/tmp/cciQsJ0d.s: Assembler messages:
/tmp/cciQsJ0d.s:30: Error: selected processor does not support `dsb 0xF' in ARM mode
/tmp/cciQsJ0d.s:37: Error: selected processor does not support `isb 0xF' in ARM mode
/tmp/cciQsJ0d.s:47: Error: selected processor does not support `dsb 0xF' in ARM mode
/tmp/cciQsJ0d.s:54: Error: selected processor does not support `isb 0xF' in ARM mode
/tmp/cciQsJ0d.s:66: Error: selected processor does not support `dsb 0xF' in ARM mode
/tmp/cciQsJ0d.s:73: Error: selected processor does not support `isb 0xF' in ARM mode
/tmp/cciQsJ0d.s:110: Error: selected processor does not support `dsb 0xF' in ARM mode
/tmp/cciQsJ0d.s:163: Error: selected processor does not support `dsb 0xF' in ARM mode
/tmp/cciQsJ0d.s:175: Error: selected processor does not support `dsb 0xF' in ARM mode
/tmp/cciQsJ0d.s:182: Error: selected processor does not support `isb 0xF' in ARM mode
32blit-stm32/CMakeFiles/BlitHalSTM32.dir/build.make:62: recipe for target '32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/main.c.obj' failed
make[2]: *** [32blit-stm32/CMakeFiles/BlitHalSTM32.dir/Src/main.c.obj] Error 1
CMakeFiles/Makefile2:140: recipe for target '32blit-stm32/CMakeFiles/BlitHalSTM32.dir/all' failed
make[1]: *** [32blit-stm32/CMakeFiles/BlitHalSTM32.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

Daft-Freak, 2020-02-01:
You should be able to upgrade WSL's Ubuntu to 18.04 with `sudo do-release-upgrade`.

Not sure what is up with your new toolchain, looks like the compiler is somehow generating the wrong code?

AndrewCapon, 2020-02-01:
The default architecture for that toolchain is probably wrong.

AndrewCapon, 2020-02-01:
Seems someone on the forum is seeing this in WSL as well!

Happy-Ferret, 2020-02-02:
> arm-none-eabi-gcc -print-multi-lib

```
arm/v5te/softfp;@marm@march=armv5te+fp@mfloat-abi=softfp
arm/v5te/hard;@marm@march=armv5te+fp@mfloat-abi=hard
thumb/nofp;@mthumb@mfloat-abi=soft
thumb/v7/nofp;@mthumb@march=armv7@mfloat-abi=soft
thumb/v7+fp/softfp;@mthumb@march=armv7+fp@mfloat-abi=softfp
thumb/v7+fp/hard;@mthumb@march=armv7+fp@mfloat-abi=hard
thumb/v7-r+fp.sp/softfp;@mthumb@march=armv7-r+fp.sp@mfloat-abi=softfp
thumb/v7-r+fp.sp/hard;@mthumb@march=armv7-r+fp.sp@mfloat-abi=hard
thumb/v6-m/nofp;@mthumb@march=armv6s-m@mfloat-abi=soft
thumb/v7-m/nofp;@mthumb@march=armv7-m@mfloat-abi=soft
thumb/v7e-m/nofp;@mthumb@march=armv7e-m@mfloat-abi=soft
thumb/v7e-m+fp/softfp;@mthumb@march=armv7e-m+fp@mfloat-abi=softfp
thumb/v7e-m+fp/hard;@mthumb@march=armv7e-m+fp@mfloat-abi=hard
thumb/v7e-m+dp/softfp;@mthumb@march=armv7e-m+fp.dp@mfloat-abi=softfp
thumb/v7e-m+dp/hard;@mthumb@march=armv7e-m+fp.dp@mfloat-abi=hard
thumb/v8-m.base/nofp;@mthumb@march=armv8-m.base@mfloat-abi=soft
thumb/v8-m.main/nofp;@mthumb@march=armv8-m.main@mfloat-abi=soft
thumb/v8-m.main+fp/softfp;@mthumb@march=armv8-m.main+fp@mfloat-abi=softfp
thumb/v8-m.main+fp/hard;@mthumb@march=armv8-m.main+fp@mfloat-abi=hard
thumb/v8-m.main+dp/softfp;@mthumb@march=armv8-m.main+fp.dp@mfloat-abi=softfp
thumb/v8-m.main+dp/hard;@mthumb@march=armv8-m.main+fp.dp@mfloat-abi=hard
```

AndrewCapon, 2020-02-02:
That looks fine to me.

Can you install this version and try it: https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2019q4/gcc-arm-none-eabi-9-2019-q4-major-x86_64-linux.tar.bz2?revision=108bd959-44bd-4619-9c19-26187abf5225&la=en&hash=E788CE92E5DFD64B2A8C246BBA91A249CB8E2D2D

Happy-Ferret, 2020-02-02:
I already use that version. That's the one that throws compatibility errors.

AndrewCapon, 2020-02-02:
ok, try 7-2018-q2-update on the same page.

Happy-Ferret, 2020-02-02:
> ok, try 7-2018-q2-update on the same page.

Same issue

AndrewCapon, 2020-02-02:
Weird, can you post on Discord exactly the steps you are taking to build?

Happy-Ferret, 2020-02-18:
I fixed it by installing a new WSL environment through lxrunoffline (quite handy for installing a new distro without trashing the old one).

closed 2020-02-18 by Happy-Ferret

alexezz21, 2021-07-02:
Include cmath and you fix it :+1: