32blit/32blit-sdk #105: Is the library avoiding C++11 on purpose?

Pharap, 2020-02-01:
There's a number of things in this library that could be improved by using C++11 features (e.g. using `constexpr float pi` instead of `#define M_PI`, using [`alignas`](https://en.cppreference.com/w/cpp/language/alignas) instead of [`#pragma pack`](https://github.com/pimoroni/32blit-beta/blob/373c7b918e143aee78df9b957eca1cc9a4125b9a/32blit/engine/utility.hpp#L7), using scoped enumerations), but before I raised any of those things as issues I wanted to check - is there a particular reason why C++11 features are being (or at least appear to be) avoided?

Gadgetoid, 2020-02-03:
Not at all- we're just used to a different paradigm of C/C++ so we're still catching up with C++11 (yes I know it's nearly a decade old at this point). We'll keep this in mind!

ali1234, 2020-02-03:
`alignas` isn't quite the same as `#pragma pack`.

Pharap, 2020-02-05:
> alignas isn't quite the same as #pragma pack.
I've checked and you're quite right, they are related but not equivalent.

In this case the usage I referred to is slated for removal so ultimately that example is probably redundant anyway.

closed 2020-08-27 by Gadgetoid