32blit/32blit-sdk #114: Bring up FAT filesystem on external Flash

Gadgetoid, 2020-02-03:
A FAT filesystem on external Flash- assuming it is possible (block size, wear leveling concerns, etc)- would be useful for myriad reasons not limited to storing games, settings, save states and Lua scripts without requiring an SD card.

It would require a new implementation of `user_diskio.c`:

https://github.com/pimoroni/32blit-beta/blob/d788e63013a37b7eac0cd19d76398844fcb624fd/32blit-stm32/Src/user_diskio.c

And linking into the FATFS system: https://github.com/pimoroni/32blit-beta/blob/master/32blit-stm32/Src/fatfs.c#L33

There's a - possibly useful - implementation available under the MIT license here: https://github.com/amvasil/stm32-qspi-fat-usb/blob/master/Src/fat_qspi_drv.c

WayneKeenan, 2020-02-04:
You may want to consider using LittleFS https://github.com/ARMmbed/littlefs

Just for external flash, not suggesting SD Card, as the one downside is that LittleFS is not lingua franca as FAT is (should 32blit support USB MSD in the future)

AndrewCapon, 2020-02-04:
How do we format, partition etc?

WayneKeenan, 2020-02-04:
It kinda depends on how the 'standard'/default model of  32blit external flash usage is envisage/designed.

In one scenario the flash could be 100% system managed and direct access is hidden from the enduser's OS File managers    e.g. there is code in the firmware to prepare LittleFS and copy files SD External to/from flash. 

In another scenario, with a bit more work, the firmware could present a FatFS via USB MSD but have it backed by LittleFS and remap on the fly.

The external flash could be split into, for example, 50% LittleFS (e.g. binaries) / 50% FatFS (e.g. assets), for convenience and if that really makes sense, lots of options.

The benefits of LIttleFS wear levelling and resiliency seem to be worth using it in some way.

WayneKeenan, 2020-02-04:
The library could be bundled with the 32blit SDK just as a optional component for the devs who go medieval, although they could obviously make it part of their own project and no need for the extra fat in 32blit code base.

It could have a 32blit specific C++ FS wrapper and it could then have a LUA binding too.

I’m rambling, just thought I’d mention it’s existence.

All the best
Wayne
(Sponsored by LittleFS :) )

Gadgetoid, 2020-02-13:
I think if we do support a filesystem on QSPI flash it's likely it will be read-only and appended as a binary blob to the end of the user code. It would be useful where user code expects a local filesystem (perhaps in a port where converting all of the assets would otherwise be painful), but wants the throughput of QSPI over that of the SD card.

Gadgetoid, 2020-03-02:
Closing for now since we've currently got our eyes on the QSPI flash for storing multiple games simultaneously and have made... some significant progress- see: https://github.com/pimoroni/32blit-beta/issues/220

I believe storing/running multiple games, and having a FAT filesystem are somewhat mutually exclusive.

closed 2020-03-02 by Gadgetoid