32blit/32blit-sdk #124: Running `make` for emscripten raises LLVM version warnings

mylogon341, 2020-02-05:
For every percentage update when running `make` on the latest master commit, the printed messages are spammed like this
```shared:WARNING: LLVM version appears incorrect (seeing "11.0", expected "10.0")
[ 63%] Building CXX object 32blit/CMakeFiles/BlitEngine.dir/types/map.cpp.o
shared:WARNING: LLVM version appears incorrect (seeing "11.0", expected "10.0")
[ 66%] Building CXX object 32blit/CMakeFiles/BlitEngine.dir/types/mat3.cpp.o
shared:WARNING: LLVM version appears incorrect (seeing "11.0", expected "10.0")
[ 69%] Building CXX object 32blit/CMakeFiles/BlitEngine.dir/types/mat4.cpp.o
shared:WARNING: LLVM version appears incorrect (seeing "11.0", expected "10.0")
[ 72%] Building CXX object 32blit/CMakeFiles/BlitEngine.dir/types/vec2.cpp.o
shared:WARNING: LLVM version appears incorrect (seeing "11.0", expected "10.0")
[ 75%] Building CXX object 32blit/CMakeFiles/BlitEngine.dir/types/vec3.cpp.o
shared:WARNING: LLVM version appears incorrect (seeing "11.0", expected "10.0")
```  
Is this a new warning, or has the target been dropped to 10.0 for some reason?

Daft-Freak, 2020-02-07:
There's possibly something wrong with your Emscripten install, I think the expected LLVM version is an internal thing that we have no control over.

mylogon341, 2020-02-07:
Ah. That makes sense. Ignore.

closed 2020-02-07 by mylogon341