32blit/32blit-sdk #142: Suggestion: Maybe need a reword in 32Blit-Loader.md

mylogon341, 2020-02-07:
Maybe just me being a bit slow, but where it says the tool is now in `tools/src/32Blit` I got a bit confused about the fact that is relative to the current path, apposed to the same directory that exists in the root directory (except that `tools/src/32Blit.cpp`) which threw me for a minute.
Feel free to close if it's not worth spoonfeeding.

Gadgetoid, 2020-02-07:
You're right- I had this same thought, but I couldn't think of a satisfactory way to clarify this to the user. My original idea was to prefix with the repository directory name, but I'm not sure that makes things better or worse!

mylogon341, 2020-02-07:
Likewise. I got to the point of `${root}/build/tools/src/32Blit` or `<root>/build/tools/src/32Blit` then realised that's actually probably much worse 😂
Maybe `./tools/src/32Blit` to show relativity? 
or maybe 
"`cd tools/src/32Blit` and now you are in the directory that ..."

Daft-Freak, 2021-05-28:
This can probably be closed as 32Blit-Loader.md no longer exists (28c8149ece9071d8127844b3ff8641d0989c5411).

closed 2021-05-28 by mylogon341