32blit/32blit-sdk #151: Let openocd select internal/external flash

ali1234, 2020-02-07:
After reset the STM32 runs whatever is in internal flash. If you are debugging something in external flash this currently means you have to go into the menu and switch.

It should be possible to use an openocd script to write a value into memory somewhere and then have the internal program read it and automatically jump to external.

This is similar to the mechanism currently used for DFU, but is a little more complicated because we have to configure QSPI first. By the time that is done, the memory address storing the boot magic has been overwritten.

Gadgetoid, 2020-02-09:
I couldn't get the backup SRAM to work, and I don't think it's the right approach for this, but I was working on formalising the reset status memory location in the linker so it would not be overwritten accidentally by other code.

The next step is to split this "register" into bits with our own meaning so we can support:

1. Reset into DFU
2. Reset into ext Flash
3. ...

Could even give half of this memory location over to the firmware to handle stuff like "Reset into USB Mass Storage Mode" or "Reset into firmware selfupdate".

```patch
diff --git a/32blit-stm32/Inc/32blit.h b/32blit-stm32/Inc/32blit.h
index 53cb8a6..4e92ccd 100644
--- a/32blit-stm32/Inc/32blit.h
+++ b/32blit-stm32/Inc/32blit.h
@@ -2,6 +2,8 @@
 #include "32blit.hpp"
 #include "fatfs.h"
 
+extern uint32_t magic_register;
+
 // Functions defined by user code files
 extern void init();
 extern void update(uint32_t time);
diff --git a/32blit-stm32/STM32H750VBTxEXT_FLASH.ld b/32blit-stm32/STM32H750VBTxEXT_FLASH.ld
index 6f21dab..283d3e4 100644
--- a/32blit-stm32/STM32H750VBTxEXT_FLASH.ld
+++ b/32blit-stm32/STM32H750VBTxEXT_FLASH.ld
@@ -47,6 +47,7 @@ LTDC (rw)           : ORIGIN = 0x2405A800, LENGTH = 150K
 RAM_D2 (xrw)        : ORIGIN = 0x30000000, LENGTH = 138K
 FRAMEBUFFER (rw)    : ORIGIN = 0x30022800, LENGTH = 150K
 RAM_D3 (xrw)        : ORIGIN = 0x38000000, LENGTH = 64K
+BACKUP_SRAM (rw)    : ORIGIN = 0x38800000, LENGTH = 4K
 ITCMISR (xrw)       : ORIGIN = 0x00000000, LENGTH = 4K
 ITCMRAM (xrw)       : ORIGIN = 0x00001000, LENGTH = 60K
 FLASH (rx)          : ORIGIN = 0x90000000, LENGTH = 32768K
@@ -192,6 +193,13 @@ SECTIONS
     *(.dma_data)           /* dma_data */
   } >RAM_D3
 
+  _magic_register_start = 0x38800000;
+  .magic_register _magic_register_start (NOLOAD):
+  {
+    . = ALIGN(4);
+    KEEP(*(.magic_register))
+  }
+
   /* Remove information from the standard libraries */
   /DISCARD/ :
   {
diff --git a/32blit-stm32/STM32H750VBTx_FLASH.ld b/32blit-stm32/STM32H750VBTx_FLASH.ld
index 42d8323..a57b7c1 100644
--- a/32blit-stm32/STM32H750VBTx_FLASH.ld
+++ b/32blit-stm32/STM32H750VBTx_FLASH.ld
@@ -47,6 +47,7 @@ LTDC (rw)           : ORIGIN = 0x2405A800, LENGTH = 150K
 RAM_D2 (xrw)        : ORIGIN = 0x30000000, LENGTH = 138K
 FRAMEBUFFER (rw)    : ORIGIN = 0x30022800, LENGTH = 150K
 RAM_D3 (xrw)        : ORIGIN = 0x38000000, LENGTH = 64K
+BACKUP_SRAM (rw)    : ORIGIN = 0x38800000, LENGTH = 4K
 ITCMISR (xrw)       : ORIGIN = 0x00000000, LENGTH = 4K
 ITCMRAM (xrw)       : ORIGIN = 0x00001000, LENGTH = 60K
 FLASH (rx)          : ORIGIN = 0x08000000, LENGTH = 128K
@@ -192,6 +193,13 @@ SECTIONS
     *(.dma_data)           /* dma_data */
   } >RAM_D3
 
+  _magic_register_start = 0x38800000;
+  .magic_register _magic_register_start (NOLOAD):
+  {
+    . = ALIGN(4);
+    KEEP(*(.magic_register))
+  }
+
   /* Remove information from the standard libraries */
   /DISCARD/ :
   {
diff --git a/32blit-stm32/Src/32blit.c b/32blit-stm32/Src/32blit.c
index cada4b5..a468859 100644
--- a/32blit-stm32/Src/32blit.c
+++ b/32blit-stm32/Src/32blit.c
@@ -35,6 +35,7 @@ extern USBD_HandleTypeDef hUsbDeviceHS;
 
 #define ADC_BUFFER_SIZE 32
 
+__attribute__((section(".magic_register"))) __attribute__ ((__used__)) uint32_t magic_register;
 __attribute__((section(".dma_data"))) ALIGN_32BYTES(__IO uint16_t adc1data[ADC_BUFFER_SIZE]);
 __attribute__((section(".dma_data"))) ALIGN_32BYTES(__IO uint16_t adc3data[ADC_BUFFER_SIZE]);
 
@@ -51,9 +52,15 @@ float volume_log_base = 2.0f;
 
 void DFUBoot(void)
 {
+  HAL_PWR_EnableBkUpAccess(); 
+  while((PWR->CR1 & PWR_CR1_DBP) == RESET){}
+  __HAL_RCC_BKPRAM_CLK_ENABLE();
+
   // Set the special magic word value that's checked by the assembly entry Point upon boot
   // This will trigger a jump into DFU mode upon reboot
-  *((uint32_t *)0x2001FFFC) = 0xCAFEBABE; // Special Key to End-of-RAM
+  //*((uint32_t *)0x2001FFFC) = 0xCAFEBABE; // Special Key to End-of-RAM
+  //magic_register = 0xCAFEBABE; // Special Key to Backup SRAM
+  *((uint32_t *)0x38800000) = 0xCAFEBABE;
 
   SCB_CleanDCache();
   NVIC_SystemReset();

```

ali1234, 2020-02-09:
What about something like this:

https://github.com/pimoroni/32blit-beta/commit/c0b6f6692d3bb45acdc924c3dd370b05537c4719

Putting it at the end of D1 means moving the stack down. Stack is ALIGN(8) so I created two 32 bit variables. Maybe that is unnecessary. They could also go in D2 or D3 I guess.

ali1234, 2020-02-09:
External flash boot check: https://github.com/ali1234/32blit-beta/commit/ab2cc8cda4f7f5920387c547ca00c0b14ea8ecdb
OpenOCD part: https://github.com/ali1234/openocd/commit/18cca6cd45156892b4bb05b10ac0f95c6ed5902b

Again, this cannot be done is `reset-end` as the firmware will have already passed this point in the boot process before OpenOCD writes the magic value. `reset-start` seems to work okay though.

Gadgetoid, 2020-03-02:
I think this has now been more formally implemented as part of the persistence API that's been shunted over to backup SRAM (not that we have a coin cell battery, but it's free RAM so we might as well use it!) see: https://github.com/pimoroni/32blit-beta/pull/236

Does this work for OpenOCD?

ali1234, 2020-03-03:
It should do. It will require some set up script, but that should not be a problem.