32blit/32blit-sdk #164: TileMap::draw only handles bounds as a multiple of 8

HeadBoffin, 2020-02-09:
I can setup world with: `TileMap world((uint8_t *) terrainTiles, (uint8_t *) terrainData, Size(32, 24), SpriteSheet::load(tilemapPackedData));`

but `world.draw(&screen, Rect(0, 0, 320, 240), nullptr);`

results in 8 rows of the 8x8 tiles drawn as expected, 8 rows of blank and then 8 rows as expected. Transforms for sprites and for the TileMap work OK. The blanks are due to the following returning -1:
https://github.com/pimoroni/32blit-beta/blob/5576e3dc53f1076e82fc829af5beae68d6742d09/32blit/graphics/tilemap.cpp#L49

I can see it's somewhere in the XOR on line 53 but I'm a bit lost on why this is here so I'm passing this on to a someone who knows the design.

This is an issue on both X & Y axes, a 12 x 8 world will have some blank columns, a 12 x 12 will have blank rows & columns. A 4 x 4 world works.

I'm using code from 8th Feb, last commit 5576e3d.

HeadBoffin, 2020-02-11:
Additional, as this seems highly related, would be very useful to have a TileMap.sprite function in the same way that Surface has because ....

I have an export script for Tiled which creates a TileMap of the world which I can use to draw a layer using the draw function, rinse & repeat for any other background layers for parallax etc, as well as attributes for a tile (solid, water, ladder etc).

I then have a smaller bespoke array of objects (again, output from Tiled) from the world Tileset that I put over the top - for instance I'm currently drawing a loverly grass area using the pirate TileMap and I want to put some logs on it here and there to demonstrate my character banging in to things.

I'm using the pirate character Tilemap for the, erm, character which is loaded in to screen.sprites.

Hence the usefulness, for me, in having a TileMap.sprite function

HeadBoffin, 2020-02-11:
Additional observation, with enough prior planning, a TileMap with everything in it is entirely possible, but as the brief is a student accessible game platform, this would ease the pain by allowing a Tileset for the world and a Tileset for the other stuff.

HeadBoffin, 2020-02-14:
Having run some tests, I totally take my hat off to lowfatcode for the sheer speed of his TileMap.draw functionality - over 20x faster than drawing each sprite from a tile map. So with a bit of documentation to highlight the benefits of having a TileMap that's a ^2 and, of course, the SpriteMap as 128 x 128 only, if we could have an official TileMap.sprite function, I think this is all good.

Gadgetoid, 2020-03-02:
You've gone over this in pretty comprehensive detail above, but to clarify to those who haven't been party to all of the talk about this on Discord- the reason TileMaps are currently limited to powers of two is because it allows for a drawing optimisation:

https://github.com/pimoroni/32blit-beta/blob/bb5e108a2ccc06b2c7d8ce9a24d879bcd0b01ba0/32blit/graphics/tilemap.cpp#L49-L61

It's Monday, so I'm not even going to attempt to explain what's going on there. But this function is called for every single pixel of a TileMap draw.

Daft-Freak, 2023-01-31:
This was marked wontfix.

I do keep coming back to the idea of a simple version of `TileMap` that is more optimised for simple use-cases (no per-line transforms) which should avoid the need for these restrictions. (If I ever write it)

closed 2023-01-31 by Daft-Freak