32blit/32blit-sdk #170: Request: Perlin noise functions.

mylogon341, 2020-02-11:
As discussed on the discord channel, this seems like a useful set of functions to have.
There are some open-source libs out there already in C++, but licensing is my concern for just using those, so I'm standing clear of this one.

nerathul, 2020-02-12:
So loaded up my perlin noise code. Created just the minimal function to get 16 values, so it will not slow down the blit. I'm still not good with C++ so if someone want to make the code more efficient just pn me on Discord so we can look together on the code. Thanks everyone who helps me with it and I hope it helps somebody.
https://gitlab.com/thetruenerathul/perlin-noise-for-32blit

Gadgetoid, 2020-02-14:
I wonder if these qualify as something that should be added to the tween functions?

https://github.com/pimoroni/32blit-beta/blob/3e7bd7396e05a8ae926e6c01bc89795096b74133/32blit/engine/tweening.cpp#L49-L72