32blit/32blit-sdk #189: Request: MIDI playback

mylogon341, 2020-02-12:
I'm not sure if this is even possible without the correct hardware or whatever but this would make music playback easy + small.

tinwhisker, 2020-02-13:
Easiest is to use a tracker.

Midi is certainly doable, but the trade-off is requiring a soundfont, which you could drop on your as card. (Might be able to convert one down small enough that you could stuff it permanently on the external flash)

Might have a play with the idea 👍

Gadgetoid, 2020-02-13:
I used DefleMask (http://www.deflemask.com/) in Commodore 64 mode for the terrible demo "song". I don't find trackers particularly intuitive versus using a piano roll and producing MIDI output, though.

I wrote the monstrosity below which extracts the raw note values from DefleMask `.dmf` files. It doesn't handle any of the instrument data, but our envelope generators and available voices align pretty closely with the SID chip so it's not hard to port across.

```python
#!/usr/bin/env python3

import argparse
import pathlib
import math
import zlib
from construct import this, len_, If, Struct, Probe, Rebuild, Const, Int8ul, Int8sl, Int16ul, Int16sl, Int32ul, Array, PrefixedArray, Computed, Switch, GreedyBytes, Enum, PaddedString, PascalString

MODE_PAL  = 985248
MODE_NTSC = 1022727
MIDI_A4 = 69
FREQ_A4 = 440.

def compute_system_total_channels(ctx):
    """Compute the number of channels in the system"""
    return {
        dmf_system.SYSTEM_GENESIS: 10,
        dmf_system.SYSTEM_GENESIS_EXT_CH3: 13,
        dmf_system.SYSTEM_SMS: 4,
        dmf_system.SYSTEM_GAMEBOY: 4,
        dmf_system.SYSTEM_PCENGINE: 6,
        dmf_system.SYSTEM_NES: 5,
        dmf_system.SYSTEM_C64_8580: 3,
        dmf_system.SYSTEM_C64_6581: 3,
        dmf_system.SYSTEM_YM2151: 13,
        dmf_system.SYSTEM_C64_8580_MODE: 3,
    }[ctx.system]

def compute_pattern_matrix_length(ctx):
    system_total_channels = compute_system_total_channels(ctx)
    return system_total_channels * ctx.rows_in_pattern_matrix

# Modes as per docs are wrong, see: http://www.deflemask.com/forum/general/dmf-specs/30/
dmf_system = Enum(
    Int8ul,
    SYSTEM_GENESIS=0x02,
    SYSTEM_GENESIS_EXT_CH3=0x12,
    SYSTEM_SMS=0x03,
    SYSTEM_GAMEBOY=0x04,
    SYSTEM_PCENGINE=0x05,
    SYSTEM_NES=0x06,
    SYSTEM_C64_8580=0x07,
    SYSTEM_C64_6581=0x17,
    SYSTEM_YM2151=0x08,

    SYSTEM_C64_8580_MODE=0x47
)

dmf_frames_mode = Enum(
    Int8ul,
    PAL = 0,
    NTSC = 1
)

dmf_instrument_mode = Enum(
    Int8ul,
    STANDARD = 0,
    FM = 1
)

dmf_note = Enum(Int16ul,
    Empty = 0,
    CSharp = 1,
    D = 2,
    DSharp = 3,
[... 148 lines cut ...]
            pitch = int(round(pitch, 0))
            output[channel] += [pitch]
            if data.note != dmf_note.Empty:
                print(f"{data.note}{data.octave} = {pitch}")

for channel in [0, 1, 2]:
    print(f"int16_t channel{channel}[{128*3}] = {{")
    for x in range(0, 128 * 3, 128):
        print(", ".join([str(n) for n in output[channel][x:x+128]]) + ", ")
    print("};")
exit(0)

for ci, channel in enumerate(result.channels):
    print(f"Channel {ci:d}")
    for pi, pattern in enumerate(channel.patterns):
        print(f"Pattern {pi:d}")
        for row in pattern.rows:
            if row.note != dmf_note.Empty:
                print(f"{row.note}{row.octave}")
            else:
                print("--")

#with open(args.out, 'wb') as output_file:
#    output_file.write(data)
```