32blit/32blit-sdk #212: Move assets pipeline from CMake into Python tooling for VS compatibility

Gadgetoid, 2020-02-17:
CMake seems to be more trouble than its worth when attempting to implement our ideal assets pipeline. It makes configuring assets difficult, and supporting new features or edge cases basically impossible.

I motion for a move to expanding the Python `asset-packer` tool into something that can manage the complete pipeline from configuration to asset output and be used both in Visual Studio, CMake and separately for prepping out-of-source binary blob assets (for loading off SD card for example).

MVP list:

- [ ] update asset-packer to understand a `config.yml` or otherwise format
- [ ] output to `assets.hpp` and `assets.cpp` (more complex projects may need to use assets across compilation units, so they shouldn't always be inlined into the header file?)
- [ ] optional output to individual packaged binary files for streaming off SD card
- [ ] multiple input files of multiple types
- [ ] multiple output files of multiple types
- [ ]  glob inputs, eg; `images/*.png`
- [ ] asset-specific arguments such as custom names, output file groups, and other conveniences

The objective should be to make `asset-packer` super simple for a beginner to pack a couple of asset files with, but extremely deep when it's required to do something more complicated.

A basic use-case should probably auto-detect most of the asset types and act accordingly:

```
asset-packer assets/*
>> found assets/sprites.png - assuming packed spritesheet
>> found assets/level.tmx - assuming Tiled level data
>> generating assets.cpp
>> generating assets.hpp
>> Use your sprites: `screen.sprites = SpriteSheet::load((const uint8_t *)assets_sprites.data)`
>> Use your level: `level = new TileMap((uint8_t *)assets_level.data, nullptr, Size(64, 64), screen.sprites);`
```

Daft-Freak, 2020-02-17:
Hmm, didn't think it was that bad, but interested to see what the new version looks like. BTW, the `(const uint8_t *)` cast is redundant.

Gadgetoid, 2020-02-17:
It certainly works, and works really well in CMake, but that doesn't translate so easily to Visual Studio (or maybe even other compile methods in future).

It's also just a little unmaintainable.

I'm currently working with an `assets.yml` that looks like this, and should satisfy all cases:

```yml
# Define an output target for the asset builder
# in this case we want a c_header (and implicitly also an object file)
# type auto-detection will notice the ".hpp" and act accordingly
assets.hpp:
  # Include assets/sprites.png
  # and place it in a variable named "sprites"
  # Since it ends in ".png" the builder will run "sprites_packed" to convert our source file
  assets/sprites.png: sprites

  # Include assets/level.tmx
  # and place it in a variable named "level"
  assets/level.tmx:
    name: level

  # Slurp up all of our .bin files and leave the auto-naming on
  assets/*.bin:

  # All of our .donkey files should be joined together
  # and placed in a variable named "donkey"
  assets/*.donkey:
    name: donkey
    join: true
```

It supports:
- Inferring output type, or explicit output types of 'hpp' or 'raw'
- Inferring input type, or explicit input types of 'sprite_packed', 'map_tiled' etc
- Single files, automatically or explicitly named
- Multiple files, automatically named
- Multiple files concatenated together and automatically or explicitly named

Gadgetoid, 2020-02-17:
/me opens `level.tmx`

`<data encoding="csv">`

:|

I could have converted this directly to Python and avoided even needing Tiled.exe installed.

FACE

PALM.

Daft-Freak, 2020-02-17:
Nothing really translates well to VS, it's why CMake exists :smile: . The new file looks nice and flexible, though it's a shame we have to lose the dependency tracking and auto-rebuilds... which is the best part of the current approach.

Daft-Freak, 2020-02-17:
Hmm.. might have some ideas to restore that. I'll wait until you push something to mess with though, as I may be completely wrong.

Gadgetoid, 2020-02-17:
There are certainly ways to restore it, at least that I can envision. I'm not convident they'll be particularly elegant though.

You can find the new Python tooling in progress at: https://github.com/pimoroni/32blit-tools

Daft-Freak, 2020-02-17:
WIP CMake integration for the new tools:
```cmake
set(32BLIT_ASSET_TOOL /home/daftfreak/.local/bin/32blit) ### FIXME

function (blit_assets_yaml TARGET FILE)
	set(FILE ${CMAKE_CURRENT_SOURCE_DIR}/${FILE})
	
	# cause cmake to reconfigure whenever the asset list changes
	set_property(DIRECTORY APPEND PROPERTY CMAKE_CONFIGURE_DEPENDS ${FILE})

	# get the inputs/outputs for the asset tool (at configure time)
	execute_process(
		COMMAND ${32BLIT_ASSET_TOOL} packer --config ${FILE} --list-inputs
		OUTPUT_VARIABLE ASSET_INPUTS
	)

	execute_process(
		COMMAND ${32BLIT_ASSET_TOOL} packer --config ${FILE} --list-outputs
		OUTPUT_VARIABLE ASSET_OUTPUTS
	)

	string(REPLACE "\n" ";" ASSET_INPUTS ${ASSET_INPUTS})
	string(REPLACE "\n" ";" ASSET_OUTPUTS ${ASSET_INPUTS})

	# do asset packing (at build time)
	add_custom_command(
		OUTPUT ${ASSET_OUTPUTS}
		COMMAND ${32BLIT_ASSET_TOOL} packer --config ${FILE}
		DEPENDS ${ASSET_INPUTS}
	)

	# add the outputs as dependencies of the project
	# TODO: assumes outputs are to be compiled
	target_sources(${TARGET} PRIVATE ${ASSET_OUTPUTS})
endfunction()

```

Usage:
`blit_assets_yaml(game assets.yml)`

All it needs is for the tool to have --list-inputs/outputs flags which just print the input/output files and exit. (I hacked something together and tested it as far as "generated rules look right"). I don't think this is too ugly?

Daft-Freak, 2020-02-19:
Getting somewhere...
```cmake
	set(32BLIT_ASSET_TOOL /home/daftfreak/.local/bin/32blit) ### FIXME

	function (blit_assets_yaml TARGET FILE)
		# cause cmake to reconfigure whenever the asset list changes
		set_property(DIRECTORY APPEND PROPERTY CMAKE_CONFIGURE_DEPENDS ${CMAKE_CURRENT_SOURCE_DIR}/${FILE})

		# get the inputs/outputs for the asset tool (at configure time)
		execute_process(COMMAND ${32BLIT_ASSET_TOOL} cmake --config ${CMAKE_CURRENT_SOURCE_DIR}/${FILE} ${CMAKE_CURRENT_BINARY_DIR}/${NAME}.cmake)
		include(${CMAKE_CURRENT_BINARY_DIR}/${NAME}.cmake)

		# do asset packing (at build time)
		add_custom_command(
			OUTPUT ${ASSET_OUTPUTS}
			COMMAND ${32BLIT_ASSET_TOOL} pack --force --config ${CMAKE_CURRENT_SOURCE_DIR}/${FILE}
			DEPENDS ${ASSET_DEPENDS}
		)

		# add the outputs as dependencies of the project (also compile any cpp files)
		target_sources(${TARGET} PRIVATE ${ASSET_OUTPUTS})
	endfunction()
```

WIP generator tool https://github.com/Daft-Freak/32blit-tools/commit/203b157fbb0deb162e5d16190f770adbf0a83ddb (Basically copy/paste of packer, then stripped down, doesn't actually list all outputs)

Gadgetoid, 2020-02-20:
I already hate the 32blit tool codebase with a passion, and I'm near determined to refactor large portions of it to make it easier to add new formats. But it's heartening that you were able to add this feature!

Gadgetoid, 2020-02-20:
I've merged your CMake command into the tool and made some fixes to ensure it outputs the right targets using all the new-style OutputFormatter stuff I smashed out today.

Reworking hasn't really changed anything of note, and the tool is in a place where I'm happy to ship a- what will end up being v0.0.2 since I reserved the pypi package name with an empty v0.0.1- version that people can start using/breaking.

My local tool locator looks like:

```cmake
find_program(
    32BLIT
    NAMES 32blit.exe 32blit
    HINTS
    $ENV{LOCALAPPDATA}/Programs/Python/Python38/Scripts/
)
```

Is there any way to make CMake search ` $ENV{LOCALAPPDATA}/Programs/Python/` or wildcard subdirs? Right now that `Python38` could be any Python 3 version and we'd want to search them all.

Alternatively if we're finding `python.exe` or `python` and we know it's version 3.x then the tool command can be `${PYTHON} -m ttblit <command_here>`

Daft-Freak, 2020-02-20:
Hmm, the existing find python command should only find python3. I don't think there's an easy way to handle all the python versions, CMake just tends to loop over ALL the versions...

Gadgetoid, 2020-02-20:
`python3` (any version) is fine for running the 32blit tool. So if we're already finding python then we can just use that and not have to try and find the tool endpoint (which is easy on *nix and a pain in the behind on Windows).

closed 2020-03-02 by Gadgetoid