32blit/32blit-sdk #221: sprite transform R90 is 1 pixel out of alignment when scaled 2x

shane-powell, 2020-02-20:
When scaled 1x these transforms look normal

```
screen.sprite(sprite, Point(0, 0), Point(0, 0), Vec2(1, 1));
screen.sprite(sprite, Point(0, 8), Point(0, 0), Vec2(1, 1), SpriteTransform::R270);
screen.sprite(sprite, Point(0, 16), Point(0, 0), Vec2(1, 1), SpriteTransform::R90);
```
[image]

However if you scale 2x the R90 transform becomes misaligned by 1 pixel

```
screen.sprite(explosionLine, Point(0, 0), Point(0, 0), Vec2(2, 2));
screen.sprite(explosionLine, Point(0, 16), Point(0, 0), Vec2(2, 2), SpriteTransform::R270);
screen.sprite(explosionLine, Point(0, 32), Point(0, 0), Vec2(2, 2), SpriteTransform::R90);
```

[image]

shane-powell, 2020-02-21:
This is also happening with the transform R180.

Daft-Freak, 2020-08-27:
Looks like this is a rounding issue, this seems to help:
```diff
diff --git a/32blit/graphics/surface.cpp b/32blit/graphics/surface.cpp
index 3173c6d..8042324 100644
--- a/32blit/graphics/surface.cpp
+++ b/32blit/graphics/surface.cpp
@@ -266,12 +266,12 @@ namespace blit {
     float bottom = sprite.h - (sprite.h - (dr.h * scale_y)) + top - 1;
 
     if (t & SpriteTransform::VERTICAL) {
-      top = sprite.h - 1 - top;
+      top = sprite.h - 0.01f - top;
       bottom = sprite.h - 1 - bottom;
     }
 
     if (t & SpriteTransform::HORIZONTAL) {
-      left = sprite.w - 1 - left;
+      left = sprite.w - 0.01f - left;
       right = sprite.w - 1 - right;
     }
 
```

PR #641 (2021-03-16): Optimise stretch_blit_sprite
Another ~9ms in blit kart, won't affect the non-scaled case (different function). I think also fixes #221 (at least, it fixed something that looked like the same bug in blit kart).

The idea here is similar to the tilemap patch: fill multiple pixels at once if scaling up.

(I was a bit surprised when I measured this, thought it had a much smaller impact...)
closed 2021-03-16 by Gadgetoid