32blit/32blit-sdk #231: SDL2 build on windows/vs2019 (32 bit) doesn't seem to "include" SDL at all.

madeso, 2020-02-27:
**Branch**: master
**Commit**: 74d97eb44eb31ad56483c4e4d98c8afe24211bd6
**CMake**: 3.16.4 (win32 bit configuration)
**Compiler**: Visual studio Enterprise 2019 (16.4.2)

**Note:**
This issue is similar to #26 but that issue seems to be "fixed", and only relate to linking to sdl main. In this case cmake "finds" sdl, but when building the compiler neither includes nor links to it or its dependencies.

**Steps to repeat**

1. Manually build SDL2 (2.0.8) and install to custom directory with the following configurations defined:
- _LIBC_: ON
- _SDL_STATIC_: ON
- _SDL_SHARED_: OFF

2. Use cmake (the gui in this case) to configure and generate 32blit-beta (root) and point
SDL2_DIR to _[sdl2-install]\cmake_

3. Build visual studio project

4. Compile errors!
```
File.cpp(11,10): fatal error C1083: Cannot open include file: 'SDL.h': No such file or directory
Input.cpp(1,10): fatal error C1083: Cannot open include file: 'SDL.h': No such file or directory
Main.cpp(9,10): fatal error C1083: Cannot open include file: 'SDL.h': No such file or directory
Renderer.cpp(3,10): fatal error C1083: Cannot open include file: 'SDL.h': No such file or directory
Audio.cpp(4,10): fatal error C1083: Cannot open include file: 'SDL.h': No such file or directory
System.cpp(4,10): fatal error C1083: Cannot open include file: 'SDL.h': No such file or directory
```

**Workaround**:
Manually add include directory and library files.

1. I hacked together a custom include dir in the root CMakeLists.txt like this:
```cmake
set(CUSTOM_INCLUDE_DIR "" CACHE STRING "custom include dirs")
include_directories(${CUSTOM_INCLUDE_DIR})
```
and pointed it to the include folder: _[sdl2-install]\include\SDL2_

2. ...and then manually added the libraries in visual studio for each executable (shmup, platformer etc).

In my case (for a debug build), the missing libraries were:

- SDL2d.lib
- SDL2maind.lib
- Winmm.lib
- Version.lib
- Imm32.lib

Daft-Freak, 2020-02-28:
Looks like there's actually two issues here:
  - Only checking for the shared target (fix: https://github.com/pimoroni/32blit-beta/pull/234)
  - SDL doesn't set the expected include directory (`[prefix]/include/SDL2`) until 2.0.9

Gadgetoid, 2020-12-30:
This issue may be long obsolete now? There have been more than a couple of changes in the intervening months.

@madeso if you have time, might be worth trying again! Closing for now, but feel free to re-open.

closed 2020-12-30 by Gadgetoid