32blit/32blit-sdk #287: 2 noob issues and fixes

mgarcia-org, 2020-03-09:
in file: https://github.com/pimoroni/32blit-beta/blob/master/docs/Linux.md

On the "sudo apt install ...." line, it should include:
python3-setuptools 

And , line "pip3 install construct bitstring" requires sudo

mgarcia-org, 2020-03-09:
The last one also on tools doc in src (updated 11 days ago):
https://github.com/pimoroni/32blit-tools/blob/master/src/README.md

Which is duplicated in root folder (updated 17 days ago):
https://github.com/pimoroni/32blit-tools/blob/master/README.md

Gadgetoid, 2020-10-01:
Every time I stick "sudo" before a command in documentation, someone drives by to say I should leave the bare command and add something like "You might need sudo." 😆 

I'm not *quite* sure how we should approach this, but it probably needs standardised somehow. Some Python setups will happily install packages for just the current user- but the root/user environments don't cross pollinate so it can - and does, in my case - confuse the heck out of people.

Oh Python 🙄 

I think you're right re `python3-setuptools` though.

Daft-Freak, 2025-01-27:
I'm going to close this as `sudo pip install` is even less likely to be the right thing to do these days...

(Though we might need a new issue for updating the tools instructions in general...)

closed 2025-01-27 by Daft-Freak

merwok, 2025-01-27:
These days, installing setuptools is mostly not required: most projects are downloaded as wheels, that don’t have build steps, and those that use sdists have metadata that specifies their build backend (setuptools or other) to the installer tool.