32blit/32blit-sdk #292: Game metadata

Gadgetoid, 2020-03-11:
At some point or another we'll probably want to append game metadata to the end of .bin files (using a hypothetical new Python tool in the 32blit-tools package - https://github.com/pimoroni/32blit-tools).

This metadata could then be read by the firmware UI to display icons, titles and maybe even small preview images.

I did some thinking about this a while back and came up with:

```python
#!/usr/bin/env python3

"""
const uint64_t blit_game_header = (uint64_t)"BLITGAME";

struct blit_game {
    const uint64_t header;
    int8ul header_version;
    rgba icon[32 * 32];
    char title[64];
    char description[256];
    char reserved[177];
    uint32_t version;
    uint16_t blocks;
};
"""

import argparse
import math
import sys
import zlib

from construct import this, Checksum, Rebuild, Padded, CString, Const, Struct, Int64ul, Int32ul, Int16ul, Int8ul, Array, GreedyBytes, Enum
from PIL import Image

image = Image.new('RGBA', (32, 32), (255, 255, 255, 0))
image = image.convert('RGBA')

asset_game = Struct(
    'header' / Const(int("".join(["{:02x}".format(ord(x)) for x in "BLITGAME"]), 16), Int64ul),
    'header_version' / Const(1, Int8ul),
    'icon' / Array(32 * 32 * 4, Int8ul),
    'title' / Padded(64, CString('utf8')),
    'description' / Padded(256, CString('utf8')),
    'reserved' / Const(bytes([0xFF] * 177)),
    'version' / Rebuild(Int32ul,
                        lambda data: 0xffffffff ^ zlib.crc32(data.data)),
    'blocks' / Int16ul,
    'data' / GreedyBytes
)

data = asset_game.build({
    'icon': image.tobytes(),
    'title': 'My 32blit game or something',
    'description': 'Some kind of description of my game',
    'version': 0x00000001,
    'blocks': 1,
    'data': bytes([0x66] * 256)
})

game = asset_game.parse(data)

print("Size: {:d}b".format(len(data)))
print("Flash pages used: {:d}".format(int(len(data) / 256.0)))
print("Flash blocks used: {:d}".format(math.ceil(len(data) / 256.0 / 64.0)))
print("Checksum: 0x{:08x}".format(game.version))
```

mgarcia-org, 2020-03-11:
it would be great to have a way to bypass the menu with a default or autostart exe.app, when first turned on.
IE, using a specifically filename in the SD root, autostart.bin etc
Then, when exiting, going back to the actual menu.

Gadgetoid, 2020-09-24:
I have started building tooling for metadata into the 32blit tool, proof of concept here - https://github.com/pimoroni/32blit-tools/pull/16

All fields are variable length and the max metadata payload size is a generous 65k.

Gadgetoid, 2020-10-01:
Closing, since we have a functional metadata proof of concept in 32blit and the tooling.

closed 2020-11-01 by Gadgetoid