32blit/32blit-sdk #313: Asset header file generation appears to be broken for small png's

ptbw, 2020-05-31:
Size of packed sprite data for small png's  appears to be wrong, header file generated shows 0x0d but size is actually 0x45. Attached is a png for a ball. Larger sprites (>255 bytes??) don't share the same issue.
(Still trying to track this down, but recording it here in case I am misunderstanding.)

[image]

**assets.yml**
```
assets.hpp:
    assets/ball.png:
        name: m_sprite_ball
        transparent: 0,0,0
```
**assets.hpp**
```
inline const uint8_t m_sprite_ball[] = {
    0x53, 0x50, 0x52, 0x49, 0x54, 0x45, 0x50, 0x4b,
    0x0d, 0x00,  // Size????
```

Daft-Freak, 2020-05-31:
So, it originally was the size of the entire asset. I (incorrectly) changed it to just the data size in https://github.com/pimoroni/32blit-tools/pull/4 while trying to fix an out-of bounds error, then the loading code was fixed to understand that in https://github.com/pimoroni/32blit-beta/pull/282 (which also fixed the out-of-bounds).

So, the small size is probably accurate and the confusion is slightly my fault.

ptbw, 2020-06-01:
So both asset creation and processing now accept the smaller data size? Please close this issue if my assumption is correct.

The reason I found the issue is that I forked a "breakout" style game (https://github.com/ptbw/32blox) and it creates assets but then does not use the API to process them. I was using it to try and get up to speed with the 32Blit.

ahnlak, 2020-06-01:
Eek, I recognise that breakout clone. Err yeah, that's in quite a mess and I was halfway through rejigging it to use the proper API stuff more thoroughly when the world got ... complicated. It might not be the best example for how the API currently works :)

Daft-Freak, 2020-06-01:
Yeah both sides should agree on the format now. (I don't think #282 is in a release yet, but you'll probably want master anyway.)

You'll have to close the issue yourself though, I'm just a contributor.

closed 2020-06-01 by ptbw