32blit/32blit-sdk #335: Boilerplate GitHub repository

Gadgetoid, 2020-08-27:
We could really use a "cookiecutter"-style method for bootstrapping new 32Blit projects.

Since each project in examples/ has a CMakeLists.txt alongside a .cpp and .hpp it would be nice to automate- other than copypasta- their creation like the original Makefile setup tried to.

Boilerplate GitHub should include:
* [ ] Visual Studio template
* [ ] Make templaye
* [ ] Functional starting set up of the asset pipeline, maybe a tilemap and sprite image
* [ ] Continuous integration setup
* [ ] Various supporting files- license, README struture, game metadata and other little things

Daft-Freak, 2020-08-27:
I have a lot of this in my repos: https://github.com/Daft-Freak/32blit-music-player (CMake + assets + Travis for device/releases)

ahnlak, 2020-08-28:
That's a great starting point, and should be easy enough to add in the VS stuff from the examples.

I'm guessing we'd also want a `init` (python?) script to ask for the location of the base 32blit install, possibly set the project name in the various makefiles... (trying to think what else I change as soon as I copy the current template!)

Daft-Freak, 2020-08-28:
I'm still not convinced about VS solutions being worth the extra work when VS has native CMake support... But, if we're going to include it in the template, why not include the VS Code setup as well.

Which would just be: (`.vscode/settings.json`)
```json
{
    "cmake.configureSettings": {
        "32BLIT_PATH": "some/path/32blit-beta",
    },
    "C_Cpp.default.configurationProvider": "vector-of-bool.cmake-tools"
}
```
(Maybe some debugger stuff, but that's a little platform specific)

ahnlak, 2020-11-11:
I was about to start adding things like assets and metadata to the `template` stuff, before I remember this idea. 

Would it be worth just quickly copying `template` into a boilerplate repo (presumably makes sense to be under `pimoroni`) and building on that?

ahnlak, 2020-11-12:
Ok, more for my own sanity than anything I've made a start on something that might be of use. It's essentially the old `template`, added assets.yml and metadata.yml, and rearranged the `CMakeList.txt` to reduce the amount of editing required.

I've not gone into any CI stuff (largely because I have no idea how any of that works...!), but it might be a useful starting point?

<https://github.com/ahnlak-blit32/32blit-boilerplate>

Daft-Freak, 2020-11-12:
Probably don't want to use `GLOB` in the template, it's not recommended by the CMake devs.

> We do not recommend using GLOB to collect a list of source files from your source tree. If no CMakeLists.txt file changes when a source is added or removed then the generated build system cannot know when to ask CMake to regenerate. The CONFIGURE_DEPENDS flag may not work reliably on all generators, or if a new generator is added in the future that cannot support it, projects using it will be stuck. Even if CONFIGURE_DEPENDS works reliably, there is still a cost to perform the check on every rebuild. 

For setting the path, you should use `-D32BLIT_PATH="/path/to/32blit/repo"` at configure ([like this](https://github.com/pimoroni/32blit-beta/blob/master/docs/32blit.md#your-own-projects))

ahnlak, 2020-11-12:
Good point on `GLOB`; I ended up using it as my source list grew too long, and I don't know cmake well enough to at least autogenerate entries for `.hpp` file as I would with old-fashioned make!

Given that the `32BLIT_PATH` is the same for all targets, does it make sense to require it on the command line instead of setting it in CMakeList?

Daft-Freak, 2020-11-12:
You can just omit headers from the source list, the only think it helps with is using IDE project generators. (which I wouldn't recommend...)

The `32BLIT_PATH` doesn't really belong in the project config, it's a property of the system you're building on. (Only really becomes a problem if multiple people are using the repo or you're using multiple machines). Even if you are specifying it on the command line, it's still a one-time setup thing... (Okay, once per target, which would probably be at most three times)

When you're using the 32blit toolchain file, you can actually omit the path... but that just just makes it harder to document :smile: 
(That was the reason the `project` line was above the line setting `32BLIT_PATH`, you generally want `project` as close to the top as possible since that's where the compiler is detected)

ahnlak, 2020-11-12:
Ahh, didn't know if it used headers for dependency scanning or anything, might not worry about that then.

The path stuff makes a lot of sense (and also means you don't have to mess about with it when you clone the repo onto a different machine!) - we should probably update the other build docs to reflect that. Does that imply that setting `32BLIT_PATH` in the file is redundant, or is it still necessary?

I moved `project` down just to logically separate out the "things you probably want to change" from the "things you probably don't" - but then if we're saying you wouldn't normally set `32BLIT_PATH` in the CMakeList, I guess we could just move that back down below `project`?

(see this is what happens when people who don't know cmake start tinkering :rofl: )

Daft-Freak, 2020-11-12:
Setting the path in the CMakeLists is mostly so that CMake GUIs show it a an option to configure, and also a default that happens to work for where the template was originally located.

The way it works with the toolchain is that it [sets `32BLIT_PATH` for you](https://github.com/pimoroni/32blit-beta/blob/master/32blit.toolchain#L48), so you end up with:
`cmake .. -DCMAKE_TOOLCHAIN_FILE=/path/to/32blit/repo/32blit.toolchain`

for a device build and:
`cmake .. -D32BLIT_PATH="/path/to/32blit/repo"`

for a local build. (`set(... CACHE ...)` won't override a value that is already set). 

Though if you're using an editor with CMake support, you can just throw the path in the settings and forget about it. ([I'm doing this with VS Code](https://github.com/pimoroni/32blit-beta/blob/master/docs/VSCode.md#cmake-arguments))

It looks like building anything other than the examples is only mentioned in the "device build" docs...

ahnlak, 2020-11-12:
I suppose I'm trying to think about how less technical users approach this; maybe rather than setting `32BLIT_PATH` to a default in CMakeList, it should just test to see if it's set (either by command line or the toolchain) and clearly say "you need to set this on your command line", rather than the current CMake-y complaint about a missing file.

It's also possible I'm overthinking this and I should not worry to much!

How does the VS stuff work, do the `sln` and `vcxproj` files need changing, or do they just key off the stuff in CMakeLists? I'm flying pretty blind on that, because these days I'm a pure Linux desktop :)

Daft-Freak, 2020-11-12:
Should be able to do:
```cmake
if(NOT EXISTS ${32BLIT_PATH}/32blit.cmake)
  message(FATAL_ERROR "...")
endif()
```

Looks like the VS files contain several `..\`s pointing to blit stuff... Not sure how that should be handled. It would also need asset support added, which is a bit more work than with CMake.

(I keep pointing out that you can just open the folder in VS and it automagically sets the project using CMake (okay, you need to tell it where SDL2 is), but some people still want the sln and stuff)

ahnlak, 2020-11-12:
Ok, tidied that up some and also realised that `project` sets the project name anyway so it's ... smaller. 

Might dive into some of the Github Action stuff in a bit, once I get some actual work done. Feel slightly guilty about cluttering up the issue here...

ahnlak, 2020-11-17:
Ok, I'm sticking a pin in it for now - CI builds now work for all platforms, and the only wrinkle is I've messed up the building of release files, but the fixes for that shouldn't interfere with anyone actually using the boilerplate itself.

I *think* as it stands it pretty much ticks all the things on your list though :)

Gadgetoid, 2020-12-11:
Thank you for all your efforts here @ahnlak 

To everyone on the sidelines, boilerplate is now here - https://github.com/32blit/32blit-boilerplate

Give it a go, raise issues over on that repo, enjoy a laughably straight forward dev experience!

closed 2020-12-11 by Gadgetoid