32blit/32blit-sdk #344: Random API inconsistencies

Daft-Freak, 2020-09-11:
Some little inconsistencies I keep noticing... then forgetting about:
 - `struct packed_image`
 - `AudioChannel::callback_waveBufferRefresh`
 - Some enum values are lowercase (`ScreenMode`, `OpenMode`, `TextAlign`), most are UPPERCASE. Looks like most of the lowercase ones are mine... Oops.
 - `blit::debug(std::string)` `blit::debugf(const char *, ...)`

Daft-Freak, 2020-09-30:
This one's in the API:
`set_screen_palette(const Pen *colours, int num_cols)` - both the implementations copy to `screen.palette`, could be replaced with something like `update_screen_palette(int num_cols)` (default to 256?). That gets screen palette updates closer to any other surface.

Gadgetoid, 2020-10-01:
Assigning @lowfatcode here (for when he gets out of the other end of the mad chaos in the warehouse) to have a little sweep over the API. Along with #354

Daft-Freak, 2021-02-22:
`TileMap`(tilemap-test, docs) and `Map` (flight, platformer, raycaster).

`Map` has layers, but uses std::vectors for storage (huge compile times for flight, more RAM/copying). MapLayer can be passed to `mode7`.

`TileMap` requires the bounds to be a power of two and can repeat/define a "default" tile.

Daft-Freak, 2021-05-28:
Ugh, I accidentally made the enum problem worse: https://github.com/32blit/32blit-sdk/blob/d7e545ddd17919fe2a190f069c38c99e0c9cb743/32blit/graphics/tilemap.hpp#L13-L16