32blit/32blit-sdk #351: Noise waveform possibly wrong?

Gadgetoid, 2020-09-17:
The noise value is re-calculated every time the waveform counter overflows to 65536.

This value should be the Be All And End All of noise, but it seems to be mangled later when it's used by the waveform generator:

https://github.com/pimoroni/32blit-beta/blob/3ddddb3e7c319ac332ebd41e1867378640e669a7/32blit/audio/audio.cpp#L80-L84

A quick tinker in Python (bear in mind I'm tired and my brain is addled so I could be spouting nonsense) suggests the output range is `-5632` to `-10751`:

```python
>>> noise = (0 & 0x0fff) - 0x07ff;
>>> sample = (noise - 0x7fff) >> 2
>>> sample += noise
>>> sample
-10751

>>> noise = (0xffff & 0x0fff) - 0x07ff;
>>> sample = (noise - 0x7fff) >> 2
>>> sample += noise
>>> sample
-5632
```

Am I wrong?

Daft-Freak, 2020-09-23:
Does looks a bit off:
[image]

(That's max volume)

Gadgetoid, 2020-09-23:
Ooof! Yeah- that’s a bit off all right! Pretty much consistent with my findings.

I wonder if it should use a LFSR with a configurable seed/taps rather than rand or blit:random. Both for consistency across platforms and - although I can’t remember the cycle timings for the HRNG - performance.

closed 2020-09-29 by Gadgetoid