32blit/32blit-sdk #376: Release Readyness Checklist

Gadgetoid, 2020-09-30:
With release hardware on the horizon we're ramping up toward a launchable codebase with wrinkles ironed out and a small wishlist of day-1 features:

1. [ ] The blit API should be reasonably stable but as long as  no. 2 is enforced this shouldn't be a huge deal, + known bugs fixed (#354 #344 #221)
2. [ ] The firmware->user code API should be FINAL (or relatively so). Changing this potentially makes binaries incompatible with the firmware.
3. [ ] We should be building/releasing firmware for Beta and Final units (merge #360 and do a test release)
4. [ ] Examples should be tidied up and converted to asset pipeline where appropriate (#365)
5. [ ] Examples should all have metadata config, icon and splash
6. [ ] PIC should be finalised and merged though I'm *keenly* aware how heavily we're leaning on @Daft-Freak for this!
7. [ ] Documentation should be given a shakeup
8. [ ] Releases should include license and documentation (#145)

Nice to have's:

1. [ ] Some method of updating firmware without DFU (#192)

Daft-Freak, 2020-09-30:
For the firmware API, I think we need a version field there and somewhere in game headers so that it can possibly be extended.

I started on an easing demo somewhere...

For PIC there are a few things left:
 - Need to fix CDC flashing (doesn't handle relocations)
 - Sort out the stdlib build, currently requires that you either build most of GCC, or needs 100MB of libs to be downloadable from somewhere. (~20MB zipped)
 - Fix the travis build
 - There's a script to prepend a header with relocation info, that probably needs something doing to it (https://github.com/Daft-Freak/32blit-beta/blob/pic-returns/tools/gather-relocs)
 - Will need a little adjusting after the metadata stuff

Should replacing the flash tool with the new one be on the list somewhere? I think it's close.

Gadgetoid, 2020-09-30:
Agreed regarding versioning the firmware API, at least that would give us some way to feed back the incompatibility rather than just crashing outright. Games would need to define a "target fw version" or something, I guess.

* We could earmark CDC flashing as a "dev only" thing and not worry too much about it for now, if it helps.
* stdlib is 😬  I guess it's easy enough just to host the libs somewhere- maybe tagged onto github releases? Guess they'd ideally need to be fetched automagically for building.
* Should `gather-relocs` be part of the 32blit Python tooling?

Added flash tool to the list! Good point.

Daft-Freak, 2020-09-30:
I don't think CDC flashing should be too hard, at worst it could just drop the relocs and flash to 0... (I haven't touched it yet, it's probably flashing somewhere weird)

If the stdlibs are hosted somewhere I think CMake has commands for downloading and extracting them. There are also the scripts I used to build them in the branch.

`gather-relocs` was a bit of a hacked together tool, not sure where it should end up... I guess every other tool for a game build is there. Its only dependency is `pyelftools`.

Gadgetoid, 2020-10-08:
As of pimoroni/32blit-tools#31 the tools will have all the pieces in place to understand the reloc structure. I'll look at getting the relocs tool merged in as part of this.

Now we have a dependency on Construct in the tools, the DFU tool can come along for the ride too.