32blit/32blit-sdk #435: Splitting "Firmware" into "Firmware" and "Launcher"

Gadgetoid, 2020-12-04:
We're running out of room for any further improvement to the firmware, so pending the merge of #388 we need to think about how the current Firmware's concerns can be separated out into:

### Firmware
* Handles the launching/execution of games
* Handles flashing games, since you can't write to flash and run from flash at the same time
* Handles core functions such as the system API- input/output, battery status, tilt, etc
* Launches the launcher by default, or - if none found in flash - finds `launcher.bin` on the SD card, flashes it and launches it

### Launcher
* Lists and launches games
* Can be substituted for any launcher capable of listing games and setting the right persistent state info
* Can be easily updated over UART with `32blit flash save`

### File Handling / File Associations
I strongly believe that the Launcher shouldn't include any specific interpreter, since it precludes other interpreters getting that same first-class treatment.

I don't see any reason why - if we are using a launcher to ask the firmware to launch a game - we cannot simply save a path to a ".lua" or ".py" or ".gb" file into persistent state, reset and have the *firmware* scan the flash for a suitable handler.

IE: the process of launching a .lua file might look like this:
1. Firmware boots
2. By default it scans flash for a "launcher" (or maybe the user picks their preferred launcher) and jumps to it
3. Launcher indexes flash and SD card, showing all supported games (either ones prefixed with metadata or with an included metadata file in their directory)
4. User selects a Lua game directory which includes "main.lua" (or whatever) and "metadata.bin" (or whatever)
5. Launcher saves `/path/to/main.lua` into persistent state
6. Reset to firmware
7. Firmware boots
8. Firmware sees `/path/to/main.lua` in persistent state
9. Firmware looks for a binary that says it can handle `lau` files and launches it
10. Lua binary runs, reads `/path/to/main.lua` from persistent state and launches game

All the non-user actions, of course, happen in a fraction of a second.

For games/files that *must* be run from Flash (.gb files for example) there's no reason these can't be flashed automatically and run like a regular file.

Daft-Freak, 2020-12-04:
There's already a way to reset to firmware (side effect of the error handling thing). So, switching game is probably as easy as:

```C++
persist.last_game_offset = ...;
api.exit(false);
```

(Would only need the header defining `persist` to be somewhere visible)

Gadgetoid, 2020-12-04:
Guess for file associations/launching files we just need a way for `last_game_offset` to be `launch_file_path`

Deleting stuff from Flash is a bit of a kerfuffle. Maybe we just need a "manage flash contents" that drops the user into a basic firmware menu that lists only internal flash for it to be deleted.

We accepted when building this thing that hardware limitations are all part of the fun, so this isn't the worst thing.

Also a `tmp` region designated in high flash for launch files and content such as roms, scripts, etc that are handled via file associations would mean these could be implicitly erased by the firmware when it needs to shove something else there, rather than explicitly managed by the user.

There's also the "if flash is just a level N cache, then do we even want to show its contents to the user/let them manage it at all" question.

Daft-Freak, 2020-12-07:
So, I had some ideas... (and worked out how to unmap flash) https://github.com/pimoroni/32blit-beta/compare/master...Daft-Freak:launch

Basically, one function exposed from the firmware: `launch(filename)`.

Which allowed me to build a simple launcher on top of my file browser:
```c++
#include "launcher-test.hpp"
#include "file-browser.hpp"

#include "engine/api_private.hpp"

FileBrowser file_browser;

void launch_game(std::string filename) {
  blit::api.launch(filename.c_str());
}

void init() {
  blit::set_screen_mode(blit::ScreenMode::hires);

  file_browser.set_extensions({".bin"});
  file_browser.set_on_file_open(launch_game);
  file_browser.init();
}

void render(uint32_t time_ms) {
  file_browser.render();
}

void update(uint32_t time_ms) {
  file_browser.update(time_ms);
}
```

Gadgetoid, 2020-12-08:
This is... beautiful. I love the poetic chaos of jumping from extflash into an API function running out of intflash... and yada yada you know the rest.

And the succinctness of the launcher code- *chefs kiss* it really does open the door for anyone to DIY if they want to add their own theme.

inb4 - https://skins.webamp.org/

Gadgetoid, 2020-12-11:
Cont from #450

This might be possible to implement without breaking back-compat:

```python
struct_blit_meta = Struct(
    'header' / Const(b'BLITMETA'),
    'data' / Prefixed(Int16ul, Struct(
        'checksum' / Checksum(
            Int32ul,
            lambda data: binascii.crc32(data),
            this._._.bin.data
        ),
        'date' / PaddedString(16, 'ascii'),
        'title' / PaddedString(25, 'ascii'),
        'description' / PaddedString(129, 'ascii'),
        'version' / PaddedString(17, 'ascii'),
        'author' / PaddedString(17, 'ascii'),
        Const(b'BLITTYPE'),
        'category' / PaddedString(17, 'ascii'),
        'filetypes' / PrefixedArray(Int8ul, PaddedString(5, 'ascii')),
        'icon' / struct_blit_image,
        'splash' / struct_blit_image
    ))
)
```

Could also go *after* icon/splash. Note the const `"BLITTYPE"` which denotes the start of a type, followed by a 17 char category (ugh why didn't I round those down), and a number of max 5-char filetypes (accepting that the 5th char is the null terminator).

Gadgetoid, 2020-12-11:
Strikes me that the *launcher* would *have* to invoke the right application, since it's going to have to know whether there's a suitable app installed for file associations to even present them to the user as runnable in the first place.

IE: in order for a launcher to turn `game.gba` into a launchable app, it must:
1. Support .gba files, and understand what they are
2. Be able to scan flash for an app that can launch .gba files

I guess an API feature would be needed to "set_launch_file" and "get_launch_file", and then it becomes a question of where the heck to stick that information so the launcher can communicate it to the uh.... uhmmm launcheee? 😆 

I guess the launch filepath could just be an optional std::string in `launch_game`/`launch_game_from_sd` that the firmware exposes via `get_launch_filename` to the launchee.

Daft-Freak, 2020-12-11:
Now I'm thinking:

- firmware starts, looks for the "launcher" and builds a list of "handlers"
- firmware starts the launcher
- launcher lists files checking if there is a handler (`api.can_launch`?)
- user starts the shiny new "GB Emulator 9000"
- now that it's flashed the firmware knows how to start `.gb` files
- user exits the emulator, or maybe "handlers" don't start like regular games
- launcher is reloaded, now shows `.gb` files as launchable, maybe with some info about what's launching it
- `launch("path/to/game.gb")`
- firmware looks up handler for `.gb` files and stores the filename as `launch_path` in persist (or something)
- emulator gets launched and calls `get_launch_path`
- game is launched

Where "launcher" is the thing that lists files and "handler" is the thing that runs them. Maybe some of that is overcomplicated... but thinking about handling "handlers" from the SD got _more_ complicated.

Gadgetoid, 2020-12-11:
Yeah it's difficult to avoid complexity either one way or another.

I like your approach because it takes some weight off the launcher and makes it easier for someone to throw one together. While it places a heck of a burden on the "firmware" maybe that's okay once the launcher has been ripped out.

We can avoid *some* complexity by blocking the install of a second launcher.

I'm trying to picture how much RAM a file association table would eat. Could easily get up to ~1k. But it'd be a heck of a lot faster than scanning the whole flash every time a folder/file is viewed.

While we're blowin' minds... PIC potentially means even a C launcher can support plugins.... 😲  but... *backs away slowly* that may be too much fun for one year.

closed 2021-01-19 by Gadgetoid