32blit/32blit-sdk #436: Mass Storage mode doesn't exit on eject in Ubuntu 20.10

Daft-Freak, 2020-12-04:
I don't know if this is just me but since updating to Ubuntu 20.10, Mass Storage doesn't exit when I "eject" the device. It seems to immediately go back to displaying "MSC mounting"/"MSC mounted" on the screen.

After ejecting sometimes the device shows up on Ubuntu as having no filesystem, sometimes the filesystem is there but not mounted. If I press eject enough times, it does get out of Mass Storage mode, but with a bunch of kernel errors.

Mostly submitting so that this is documented in case anyone else runs into it. (I can't really debug as Mass Stoarge mode causes my desktop to hang for a few seconds on mount)

(blank lines whenever I tried to eject)
```
[77460.930044] usb 1-4: new full-speed USB device number 94 using xhci_hcd
[77461.270791] usb 1-4: New USB device found, idVendor=0483, idProduct=572a, bcdDevice= 2.00
[77461.270794] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[77461.270795] usb 1-4: Product: 32Blit Mass Storage
[77461.270797] usb 1-4: Manufacturer: STMicroelectronics
[77461.270797] usb 1-4: SerialNumber: 305139533437
[77461.285817] usb-storage 1-4:1.0: USB Mass Storage device detected
[77461.285977] scsi host10: usb-storage 1-4:1.0
[77462.317303] scsi 10:0:0:0: Direct-Access     STM      32Blit SD        0.01 PQ: 0 ANSI: 2
[77462.317593] sd 10:0:0:0: Attached scsi generic sg4 type 0
[77462.317997] sd 10:0:0:0: [sde] 3911680 512-byte logical blocks: (2.00 GB/1.87 GiB)
[77462.318237] sd 10:0:0:0: [sde] Write Protect is off
[77462.318239] sd 10:0:0:0: [sde] Mode Sense: 00 00 00 00
[77462.318485] sd 10:0:0:0: [sde] Asking for cache data failed
[77462.318488] sd 10:0:0:0: [sde] Assuming drive cache: write through
[77462.384278]  sde: sde1
[77462.386053] sd 10:0:0:0: [sde] Attached SCSI removable disk
[77467.707943] FAT-fs (sde1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.

[77485.414090] sd 10:0:0:0: [sde] Spinning up disk...
[77486.441897] .ready
[77486.459180]  sde: sde1

[77543.719653] sd 10:0:0:0: [sde] Spinning up disk...
[77544.745581] .ready
[77544.762706]  sde: sde1

[77568.920357] sd 10:0:0:0: [sde] Spinning up disk...
[77569.929462] .ready
[77569.946771]  sde: sde1

[77575.178915] usb 1-4: USB disconnect, device number 94
[77575.179157] sd 10:0:0:0: [sde] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK cmd_age=0s
[77575.179161] sd 10:0:0:0: [sde] tag#0 CDB: Read(10) 28 00 00 3b ad 81 00 00 f0 00
[77575.179163] blk_update_request: I/O error, dev sde, sector 3911041 op 0x0:(READ) flags 0x84700 phys_seg 30 prio class 0
[77575.179189] blk_update_request: I/O error, dev sde, sector 3911281 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 0
[77575.179247] blk_update_request: I/O error, dev sde, sector 3911041 op 0x0:(READ) flags 0x0 phys_seg 8 prio class 0
[77575.179251] Buffer I/O error on dev sde1, logical block 3910912, async page read
[77575.179254] Buffer I/O error on dev sde1, logical block 3910913, async page read
[77575.179255] Buffer I/O error on dev sde1, logical block 3910914, async page read
[77575.179256] Buffer I/O error on dev sde1, logical block 3910915, async page read
[77575.179257] Buffer I/O error on dev sde1, logical block 3910916, async page read
[77575.179258] Buffer I/O error on dev sde1, logical block 3910917, async page read
[77575.179259] Buffer I/O error on dev sde1, logical block 3910918, async page read
[77575.179260] Buffer I/O error on dev sde1, logical block 3910919, async page read

```

Gadgetoid, 2020-12-07:
I get the same hang with Windows 10 using Mass Storage Mode. It's definitely a little quirky. I think I have 20.04 on my Linux desktop so I can try it there, too.

Daft-Freak, 2020-12-24:
Hmm, eject works fine on a pi...

Daft-Freak, 2021-06-10:
Seems eject is working fine again on 21.04. So maybe an Ubuntu/kernel bug?

Daft-Freak, 2022-01-08:
... and not working again (same "Spinning up disk..."). Running 21.10 now. Wonder if it still works on the pi with bullseye...

Daft-Freak, 2022-10-27:
Looks like #774 fixed this. (One of my blits was working and the other wasn't... so I bisected it to that change)

closed 2022-10-27 by Daft-Freak