32blit/32blit-sdk #440: Launching a game when nothing is selected.

shane-powell, 2020-12-08:
The firmware will try to launch a game even if nothing is selected in the list. This throws up the restart game window and if you press yes it will freeze the UI.

Gadgetoid, 2020-12-09:
To throw into the mix here, it also implodes when a game is *deleted* when nothing is selected, or deleted twice somehow.

Had some pretty wonky blits for a moment when this caught me out.

Daft-Freak, 2020-12-09:
Yeah, not checking that the game list is non-empty then reading out-of bounds:

https://github.com/pimoroni/32blit-beta/blob/1e7ed87f8413a6ef434f1931a8a5c71abc1a3a14/firmware/firmware.cpp#L510-L513

https://github.com/pimoroni/32blit-beta/blob/1e7ed87f8413a6ef434f1931a8a5c71abc1a3a14/firmware/firmware.cpp#L536-L538

Err... oops (also, not the most consistent file)

a2, 2020-12-29:
@Gadgetoid I think we might be able to close this issue?

closed 2020-12-29 by Gadgetoid