32blit/32blit-sdk #461: SD timeout handling is broken

Daft-Freak, 2020-12-14:
Most of the SD code handles timeouts like this:
```c++
Timer1 = 200;
do {
	// do thing
} while(Timer1);
```
(or `Timer2`)

... there isn't any code to decrement either of those. A few places use `HAL_GetTick`, which works most of the time, but the MSC code does calls from an interrupt... so that doesn't work either.

Gadgetoid, 2020-12-20:
Looks like it was intended for these values to be decremented via the SysTick_Handler which is... fair enough I suppose.

Fix should be something like:

```patch
diff --git a/32blit-stm32/Inc/fatfs_sd.h b/32blit-stm32/Inc/fatfs_sd.h
index 41f233c..6c0f52c 100644
--- a/32blit-stm32/Inc/fatfs_sd.h
+++ b/32blit-stm32/Inc/fatfs_sd.h
@@ -3,6 +3,8 @@
 #include "ff.h"
 #include "spi.h"
 
+extern void SDTimer_Handler();
+
 /* Definitions for MMC/SDC command */
 #define CMD0     (0x40+0)     	/* GO_IDLE_STATE */
 #define CMD1     (0x40+1)     	/* SEND_OP_COND */
diff --git a/32blit-stm32/Src/fatfs_sd.c b/32blit-stm32/Src/fatfs_sd.c
index 9e9d202..d979039 100644
--- a/32blit-stm32/Src/fatfs_sd.c
+++ b/32blit-stm32/Src/fatfs_sd.c
@@ -13,6 +13,11 @@ static volatile DSTATUS Stat = STA_NOINIT;	/* Disk Status */
 static uint8_t CardType;                    /* Type 0:MMC, 1:SDC, 2:Block addressing */
 static uint8_t PowerFlag = 0;				/* Power flag */
 
+ void SDTimer_Handler() {
+	 if(Timer1 > 0) Timer1--;
+	 if(Timer2 > 0) Timer2--;
+ }
+
 /***************************************
  * SPI functions
  **************************************/
diff --git a/32blit-stm32/Src/stm32h7xx_it.c b/32blit-stm32/Src/stm32h7xx_it.c
index f655941..991bb1b 100644
--- a/32blit-stm32/Src/stm32h7xx_it.c
+++ b/32blit-stm32/Src/stm32h7xx_it.c
@@ -22,6 +22,7 @@
 #include "main.h"
 #include "stm32h7xx_it.h"
 #include "fatfs.h"
+#include "fatfs_sd.h"
 #include "gpio_defs.h"
 /* Private includes ----------------------------------------------------------*/
 /* USER CODE BEGIN Includes */
@@ -200,6 +201,7 @@ void PendSV_Handler(void)
 void SysTick_Handler(void)
 {
   /* USER CODE BEGIN SysTick_IRQn 0 */
+  SDTimer_Handler();
 
   /* USER CODE END SysTick_IRQn 0 */
   HAL_IncTick();
```

Daft-Freak, 2020-12-20:
Hmm, I guess... should probably be `volatile` as well. Still doesn't help MSC though since it's called from the OTG interrupt.

I wonder how many of these timeouts are actually useful... if we're the master, can a single transfer time out?

Gadgetoid, 2021-01-19:
I think it's safe to say you've nailed this one- I did literally ping my SD card across the room trying to get it to fail to mount.

closed 2021-01-19 by Gadgetoid