32blit/32blit-sdk #470: Graphical corruption when installing a game with no metadata

ali1234, 2020-12-18:
[image]

There should be some kind of fallback to prevent this from happening. Either in the launcher, or by providing default metadata during build if metadata.yaml is not present.

ahnlak, 2020-12-18:
I would say in the launcher (at least); feels risky to rely solely on the build

ali1234, 2020-12-18:
Agree, but it may not be possible depending on how the metadata is formatted inside the binary.

Daft-Freak, 2020-12-18:
There's a sort-of fallback now in that the metadata is blank and the title is "game@[block]". Could certainly be improved a bit though.

ali1234, 2020-12-21:
I just realised that my screenshot doesn't show corruption. That is just what "@" looks like in the tiny font.

closed 2020-12-21 by ali1234