32blit/32blit-sdk #471: Please document final button layout

ali1234, 2020-12-18:
I looked at production units on twitter and it seems that menu and home have swapped places since beta.

Also docs/32blit.md seems to contradict itself in a few places about which button to press, but it is difficult to know what is right and wrong without a canonical reference for the buttons.

ali1234, 2020-12-18:
This is what we've worked out so far:

On release units:

 * Home button is left, and accesses the system menu. Long press exits game.
 * Menu button is right, and is for use by games.

On beta units, the labels for Menu and Home buttons are swapped, but they are electrically the same. The left button should open the system menu, even though it has the Menu icon.

Gadgetoid, 2020-12-20:
To pour petrol on the fire of confusion here- my unit has the "Home" on the left, and "Menu" on the right since it falls somewhere between Beta and Release stages. However it is the *right*-most button which exits games and accesses the system menu.

This button has an external interrupt, so in early firmware- before we had the concept of an API and the firmware handling all sorts of system IO exclusively - it served as an entry point from games running on external flash into the firmware.

It also does some gymnastics with a timer to handle the long press.

None of this stuff is really relevant in the new model, where the firmware owns the system IO and communicates it to the user apps via the "API" so we're probably well overdue assigning that button to "Short Press -> System Menu" and "Long Press -> Exit Game".

## AB/XY

The A, B, X and Y buttons are intended to correlate with those on the SNES controller- including loosely matching the colour. IE:

* Bottom = B, Yellow
* Right = A, Red
* Top = X,  Blue
* Left = Y, Green

I'd use something like an N30 Pro 2 controller for testing games on a desktop... at least until someone whips up a USB controller firmware for 32blit itself.

ahnlak, 2020-12-22:
Just to add to this; I *think* the API currently mislabels HOME and MENU, in that the menu button (probably on the right, and to be used by games) is Button::HOME, and the system button is Button::MENU.

We probably want to change that sooner rather than later, as it's something of a breaking change (and I doubt hundreds of games are yet using it...)

lenardg, 2020-12-22:
My beta unit physical buttons: LEFT: home, RIGHT: menu
My retail unit physical buttons: LEFT: home, RIGHT: menu

However, what the buttons do ...

My beta unit: LEFT button invokes system menu. Long press exits the game.
My retail unit: RIGHT button invokes system menu. Long press exits the game. 

Just so it wouldn't be that straight forward :D