32blit/32blit-sdk #501: Turn display off

wdevore, 2020-12-28:
Not exactly an issue but more of a request.

Would it be possible to have a menu option for turning the display off in the next release?

I ask because the only way to turn it off is when I unplug it, but then the battery starts to discharge.

Thanks.

lenardg, 2020-12-28:
Battery drain while the unit is off seems to be very minimal. I have had my 32blit here in "off" mode, charged it about a week ago, and it is still full, almost no battery is gone (I turned it on for maybe a few minutes during this week, I mostly use my Beta unit for development). 

So I am wondering why would you need to have a mode, where it is plugged in with the display off?

mikerr, 2020-12-28:
You can always select DFU mode of you want the screen off ?
Just press reset to bring it back on ...

wdevore, 2020-12-28:
Aren't TFT susceptible to burn in? My unit drains the battery 30% in 1 day with it powered off.

Gadgetoid, 2020-12-28:
"off" is effectively "shipping mode", and is the low power mode that the charge controller is put into in order to ship the unit with a LiPo battery installed. Because it- for lack of a better term- "disconnects" the battery in lieu of turning the system off the alternate power source over USB will keep 32blit powered on.

We need to investigate a low power "Sleep" mode and possibly also a "Display Sleep" mode that turns off the display after a period of inactivity.

Gadgetoid, 2020-12-29:
Tried to shunt the display controller into sleep mode today, but it wasn’t cooperating. Possibly continuing to send data is something of a mixed message.

Did enough of a cursory research of the process of sleeping/waking the STM32 also. Enough to scare myself off until things are a little less hectic 🤣

ahnlak, 2021-03-19:
Is this not more-or-less covered by https://github.com/32blit/32blit-sdk/pull/577 (and it's followups)?

Daft-Freak, 2021-03-19:
I think that covers "display sleep" but not "real" sleep, then again this issue was about turning the display off which you can sort-of do now... if you wait two minutes.

Also, my "other" blit has been sitting in "Power Off" for over a week and has lost about half of it's battery, which is... surprising (I thought almost everything was off in that case?). Based on the other comments here it seems quite variable how long the battery lasts in "Power Off"...

Gadgetoid, 2021-03-20:
AFAIK "Power Off" is about as off as it can get- IE it's the "Shipping Mode" of the onboard charge controller (bq24295) and is the state the system is required to be in for shipping since everyone is terrified of LiPo batteries these days.

It still has *some* quiescent or shutdown current presumably to power whatever logic is required to turn the system back on- in fact the datasheet (https://www.ti.com/lit/ds/symlink/bq24295.pdf) lists this as 20μA max an 15μA typical so... yeah, actually draining 50% of the LiPo cell capacity while off feels weird. That said I can't *quite* vouch for the accuracy of the system battery meter so it might just be a "blit is lying to you" issue rather than leakage current. Or both. The self discharge rate of LiPos is pretty low, I think, so that wouldn't explain it.

My batteries are always dead since I never turn my blits off :laughing:

Daft-Freak, 2021-03-21:
Hmm, *adds logging and leaves blits on*...

[FOUR HOURS LATER]

[image]

(The Y axis is the range of the battery meter)

ahnlak, 2021-03-21:
After leaving 'hardware test' running, I also noticed that sleep doesn't turn off the LED...

Gadgetoid, 2021-03-21:
> (The Y axis is the range of the battery meter)

:grimacing:

Daft-Freak, 2021-04-11:
> Also, my "other" blit has been sitting in "Power Off" for over a week and has lost about half of it's battery, which is... surprising (I thought almost everything was off in that case?). Based on the other comments here it seems quite variable how long the battery lasts in "Power Off"...

So... after some "extensive testing" (haven't used one of my blits since my last comment here), I think I may have misremembered my battery level previously. This time I've gone from 4.0v to... 4.0v.

closed 2025-01-27 by Daft-Freak