32blit/32blit-sdk #506: Tween behaviour with non zero starting floats.

Loxrie, 2020-12-29:
Is it tweens intention to always start from 0.0f and progress towards 1.0f?

If I initialise tween like so, for example:

`
sprite_tween.init(tween_linear,152.0f, 180.0f, 60, 1);
sprite_tween.start()
`

Then the first "tic" of my tween in an update or render can produce a value of 0 as "from" is not copied to "value" in init() or start().

I was hoping to use it for controlling speed of some sprite animations fixed by 8 pixel widths, so i could use it for simulating trudging or sliding through terrain but it doesn't seem suitable as it's not guaranteed to start at my from value, and it's not guaranteed to reach my destination before finishing.

ahnlak, 2020-12-29:
I definitely works across different ranges - I use them a fair bit. You're right, though, that `start()` should probably set a default value.

Loxrie, 2020-12-29:
Excellent good to know I wasn't off the mark.  I did try forcible setting value myself but didn't have much luck.

closed 2020-12-29 by Loxrie