32blit/32blit-sdk #512: Should it be possible to start a Timer again in the callback?

Loxrie, 2020-12-30:
In my TimerCallback I've tried using the Timer param, and an external pointer reference, to reinitialise the timer and start it but it never fires, for example:

```c++

std::vector<ghostState> cycleStates = {
  ghostState::CHASE,
  ghostState::SCATTER,
  ghostState::CHASE,
  ghostState::SCATTER,
  ghostState::CHASE,
  ghostState::SCATTER,
  ghostState::CHASE
};

std::vector<int8_t> cycleTimes = {
  20,
  7,
  20,
  5,
  20,
  5,
  -1
};

int8_t blinky_cycle_index = 0;
int8_t pinky_cycle_index = 0;
int8_t inky_cycle_index = 0;
int8_t clyde_cycle_index = 0;

void manage_ghost_move_state(Timer &t, Ghost *g, int8_t *cycle_index) {
  ghostState nextState = cycleStates[*cycle_index];
  printf("manage_ghost_move_state setting state %d for ghost %s\n", nextState, g->name.c_str());
  g->set_move_state(nextState);
  if (*cycle_index < cycleTimes.size()) {
    int8_t next_cycle_time = cycleTimes[*cycle_index];
    if (next_cycle_time != -1) {
      // Setup next ghost cycle.
      printf("manage_ghost_move_state %s timer pre  dur - %d started - %d loops - %d state - %d\n", g->name.c_str(), g->move_cycle_timer.duration, g->move_cycle_timer.started, g->move_cycle_timer.loops, g->move_cycle_timer.state);
      g->move_cycle_timer.duration = cycleTimes[*cycle_index] * 1000;
      g->move_cycle_timer.loops = 1;
      g->move_cycle_timer.start();
      printf("manage_ghost_move_state %s timer post dur - %d started - %d loops - %d state - %d\n", g->name.c_str(), g->move_cycle_timer.duration, g->move_cycle_timer.started, g->move_cycle_timer.loops, g->move_cycle_timer.state);
      cycle_index++;
    }
  }
}

void blinky_cycle_timer_callback(Timer &t) {
  printf("blinky::move_cycle_timer_callback fired %d.\n", blinky_cycle_index);
  manage_ghost_move_state(t, blinky, &blinky_cycle_index);
}
```

I think this is because the TimerCallback is executed prior to the transition to FINISHED, so my code resets duration and loops then calls start only for the Timer code itself to immediately mark my timer as finished.

I don't know if this is intentional or if there are any knock on effects from modifying this behaviour?  My hope was a simple way of increasingly changing object behaviour over time.

Loxrie, 2020-12-30:
Are you guys happy for me to attach proposed PRs?  Is there a format or template you use for this?

This solves my problem, but I was concerned with regards to knock on effects for infinite timers:

https://github.com/pimoroni/32blit-beta/compare/master...Loxrie:feature/fix-512?expand=1

Gadgetoid, 2020-12-30:
Could you use `g->move_cycle_timer.loops = -1` so that the timer always implicitly loops and restarts? I don't think there's any danger in changing the duration on the fly in this case.

Loxrie, 2020-12-30:
I sure can, I'll try that, then I just have to stop it when I run out of cycles.

Gadgetoid, 2020-12-30:
That said - I don't  think there are any knock-on effects to moving the callback *after* the loop decrement. You could argue that this better reflects the number of remaining iterations and the timer state at the time the callback is fired.

Right now you can never know- from callbacks alone- if the timer has *actually* finished, without doing some weird gymnastics.

In case of your tweak, just moving the callback right after the if/else block should be fine.

This *might* have knock-on effects in user code, but I think your way is the right way.

Loxrie, 2020-12-30:
Your suggestion also works. 👍 

Should I submit that PR?

Gadgetoid, 2020-12-30:
Yup just stick the callback after the `if/else` and raise a PR. I'll see if it breaks anything :laughing: and then warn people with the next release.

Pause might be useful. You're welcome to try!

Loxrie, 2020-12-30:
Ok I've moved the callback so there's only one of them and pushed it to the feature fix branch.  I'll raise that PR.

PR #513 (2020-12-30): Feature/fix 512 - Moved location of TimerCallback execution.
Proposed Fix #512 

TimerCallback is now executed after t->started set for infinite timers, or after timer has transitioned to FINISHED state for non infinite timers.

This allows the timer callback function to change timer params and restart the timer without it immediately transitioning to FINISHED.

This could be a breaking change for other developers.
closed 2020-12-30 by Gadgetoid