32blit/32blit-sdk #544: 1.7-Beta broken

wdevore, 2021-01-10:
I upgraded from **1.5beta** to **1.7beta** and now I can't launch any of the games with "A" button. Actually none of the button keys work except "Menu". And I also don't see the mentioned contributers scroll thingy either.

I did the usual of copying the launcher on to my sd and the Menu says I have 1.7 installed.

Any ideas?

mikerr, 2021-01-10:
You'll get "dead keys" like that using 0.15 launcher on newer firmware 

The launcher doesn't auto-update just by copying a new onto SD (maybe it should), you have to flash it.

E.g on Linux:
`sudo 32blit flash flash --file=launcher/launcher.blit`

ahnlak, 2021-01-10:
I think we need to improve the upgrade process; since the firmware / launcher split things have become ... complicated.

Daft-Freak, 2021-01-10:
Hmm, the firmware would need to know the file on the SD card is newer (metadata version is always "v1.0.0"). I guess there's the datetime field, or it could just delete the file after flashing it.

Gadgetoid, 2021-01-10:
The problem here is an API break- the lack of input in the Launcher is due to the Firmware and Launcher disagreeing about where in RAM the button states are stored. This should be less common as the 32blit software matures.

Assuming people might copy `launcher.blit` and `firmware-update.blit` to their SD card and run one or the other first, we need some means for:

a. Forcing the firmware to run a `firmware-update.blit` from SD card
b. Forcing the firmware to install a `launcher.blit` from SD card

I think always installing `launcher.blit` if it exists is a good idea, though it would have to be moved/deleted after installation. The logical gotcha here is that in the above case, the 32blit will boot up, install the incompatible *launcher* and leave you with no way to update the *firmware*.

So I think a modifier button should have to be held down in order to force a reinstall of launcher.blit where a launcher is already installed... I appreciate we're overloading the system with button combos and this might not be particularly intuitive, but I can't see a way around this?

The same, perhaps, should apply to `firmware-update.blit`. If it exists and "some button" is held down, then run it.

Thoughts?

Daft-Freak, 2021-01-10:
Could make the firmware check for `firmware-update-*` first. Flash/delete/run that, then it would reset and do the launcher update automatically.

wdevore, 2021-01-10:
I tried running 32blit to update the launcher, but for some reason 32blit can't be found "```RuntimeError: Unable to find 32Blit```" when using:
```
sudo 32blit flash flash --file=launcher/launcher.blit
```
Yet I can type "32blit" and it is present on my system.

Any thoughts?

wdevore, 2021-01-10:
@Gadgetoid I can't run the ```firmware-update.blit``` file because none of the buttons work and I tried @mikerr 's suggestion but 32blit can't be found.

For now I have reverted back to--and am stuck at-- **1.5beta**. **1.6beta** has the same issue.

I sure HOPE that **1.8beta** has a fix for this. Here to "waiting".

Cheers

Daft-Freak, 2021-01-10:
Okay, that message in the tool may need adjusting, it's referring to the hardware, not the identically named tool...

Anyway, if anyone feels like testing [here's a slightly updated firmware/updater](https://github.com/pimoroni/32blit-beta/files/5792988/fw-upd.zip). Just put launcher.blit on the SD card and DFU the firmware.bin/.dfu file. (It's the 0.1.7 launcher)

(That firmware can also install firmware-update.blit... but you would need the update installed to do that...)

wdevore, 2021-01-10:
So I copy only the *launcher.blit* file to the SD card and not the *firmware-update.blit* file?

Nonetheless, i copied the launcher and ran dfu-util and that fixed it. :+1:  :heavy_check_mark:

Gadgetoid, 2021-01-10:
Looks like Charlie's in the business of granting wishes tonight, since #545 should hopefully avoid any future sticky firmware/launcher interdependence mess and generally makes the process of updating both - via an SD card - really slick.

I'm going to tag a new release ASAP with this change and the handy fixes to surface.

Might have to *downgrade* my 32blit to launch and make sure the process works as I'd expect. Sometimes the updater falls afoul of API changes too.

Thank you @Daft-Freak

wdevore, 2021-01-12:
Also, 1.8beta fixed it too.

closed 2021-01-12 by wdevore