32blit/32blit-sdk #550: Question: Recommend way to exit a game back to menu

wdevore, 2021-01-14:
Currently I'm using ```exit(0)``` which I'm sure isn't the best way.

What is the recommended way of returning back to 32Blit menu?

ahnlak, 2021-01-14:
My instinct would be to say that the 'recommended way' is for the user to use the normal system menu option 'Exit Game'.

Gadgetoid, 2021-01-14:
On the 32blit it would be the private API function `blit::exit(bool is_error)` which will throw up an error message prompting you to restart the game if you pass `true` or just reset back to firmware->launcher if you pass `false`.

I believe you can access this by using:

```c++
#include "engine/api_private.hpp"
```

Normally, however, you'd just let the user exit by long pressing the Home button.

Daft-Freak, 2021-01-14:
... and `exit` is implemented on the blit as `blit::api.exit(code != 0);`. (That was implemented so that `abort` would exit the game, but makes exit do the "right" thing in other cases too...)

https://github.com/pimoroni/32blit-beta/blob/9033cdfbcac19cc0df66caab3c9b7728f3f3a9c6/32blit-stm32/startup_user.cpp#L22-L24

wdevore, 2021-01-14:
Excellent info.

closed 2021-01-14 by wdevore